rwaldron/johnny-five #861: How to reset all pin states?

virgilvox, 2015-07-11:
I'm building some code that does dynamic stuff. There is a config event that happens that reconfigures pin states based on some incoming JSON. I'd like to reset all the pins so that there are no pin conflicts in the event a pin's config is now a Write instead of a Read.

Is there a method for resetting all pin states built into johnny-five?

rwaldron, 2015-07-11:
Take a look at this issue https://github.com/rwaldron/johnny-five/issues/617 

- Are all the pins instances of `Pin`? 
- If you have control over all the pins, just initialize them once and update the mode as new configuration comes along. 

```js
var pin = new five.Pin(13);

// sometime later...
pin.mode = five.Pin.MODES.OUTPUT;

// even later...
pin.mode = five.Pin.MODES.INPUT;
```

virgilvox, 2015-07-11:
Thanks for the quick reply Rick!

Would it then work if ..

```
//at first
  board.pinMode(5, five.Pin.INPUT);

board.digitalRead(5, function(value) {
                debug(value);
              });

//then later..

  board.pinMode(5, five.Pin.OUTPUT);

```

Does the event handler remove itself?

virgilvox, 2015-07-11:
Also, there is the case that the configuration set will have removed a config for a pin and that would need to get disabled. 

My code https://github.com/octoblu/dynamic-j5/

Resseguie, 2016-01-23:
This is something @monteslu has also spent some time thinking about.

reconbot, 2016-05-25:
I'm going to close this issue due to it's age, but if you'd like to continue with it feel free to reopen. I know it's not something J5 really supports right now.

closed 2016-05-25 by reconbot