rwaldron/johnny-five #865: i2c missing start bits for higher addresses on Edison

dreyna7399, 2015-07-13:
I installed Johnny-Five on my Intel Edison with the Intel Edison Kit for Arduino .2 and the Grove shield this last May. I am finding that I cannot write to I2C devices with higher addresses because the start bit is missing.

I tried to add a Grove Barometer I2C device (at address 0x77) but could not communicate with it. Looking at the signals with a logic state analyzer I could see that the clock started without a start bit, so there were no proper i2c frames.

Looking again at my 'working' Grove LCD display from the Johnny-Five examples, I found that while the character writes to address 0x3e were clean, the backlight write to address 0x62 had the same issue of clocks with no start bit. I also realized that the backlight was only 'working' because it was somehow able to adjust just enough to 'mostly' set the values, though hard testing showed that it could take several writes to truly set the backlight correctly.

It appears to me that there is an error in the i2c library when writing to addresses higher that 0x3e. Both devices had completely clean i2c traces when I put the shield on an Arduino Uno and ran the tests from there, so the issue seems to be either Fermata and/or Edison related.

dreyna7399, 2015-07-14:
1. The attached "bad_i2c_0x62_edison_j5.png" shows the broken i2c waveform on my Edison+Johnny-5.

2. The attached "good_i2c_0x62_set_arduino.png" shows the working i2c waveform on my Arduino.

As you can see that (ignoring the gliches on the data line) it is the same waveform except for the beginning, where the data line does not go down to form the start bit..

3. I have debugged it down to "node_modules/edison-io/node_modules/galileo-io/lib/galileo.js"

  591  if (buffer.length) {
  592    state.i2c.address(address);
> 593    state.i2c.write(buffer);

Where:
  0: address = 98
  1: cmdRegOrData = 4
  2: inBytes = [128]
  3: arguments = {"0":98,"1":[4,128],"length":2,"callee":"function (...
  4: buffer = {"0":4,"1":128,"length":2,"parent":...}

I cannot debug it farther in nodejs since "state.i2c.write" appears to be a C module.

---

[image]

[image]

dreyna7399, 2015-07-15:
Hmm, I am new to J5 but it appears that "state" is the J5 Edison board, "i2c" is the MRAA interface, and "write" is "mraa_i2c_smbus_access" (which merely shepherds the request to an IOCTL), implying that this is a defect in the kernel driver for i2c. 

Not what I was expecting or hoping. I am now on to Yocto Project and kernel.org.

rwaldron, 2015-07-15:
Thanks for the report! 

Let's keep this open and as you learn more, you can update here? I think that would be great for anyone that encounters the same issue in the future.

dreyna7399, 2015-07-21:
I have been debugging this, and this is what I have so far with my attempts at a simple reproducer: The "LCD" refers to the Grove LCD with RGB backlight (version 2.0).

1. I have had runs with zero errors with my J5 LCD code and my standalone minimal python code version, and then runs with 5% to 80% errors. I have not found a pure set of events to lead to the failure yet (but see below).

  * I currently suspect that poor i2c devices can poison the fragile i2c multiplexed GPIO implementation. 

  * I also suspect that the i2c bus float time while it waits for an init (which can be several minutes) may allow false states.

  * None of the other common boards (like BeagleBone, MinnowMax, nor WandBoard) use "CONFIG_I2C_GPIO=y" so I cannot do comparisons that way.

  * The errors are manifest with half-width start clocks and the data line not starting low to form the start bit. You also can see the errors in strace as "EREMOTEIO (Remote I/O error)".

2. I2C devices that poison the bus

  * I do observe that the Grove LCD will definitely  poison the i2c buss if you do a read on it, for example with "i2cdetect". When its "0x3e" address is read, the data line stays high for the 8 bits, then somehow goes low and stays low without allowing for a stop bit until you get start getting lost arbitration errors and i2c-designware-pci backtraces. After that the i2c bus unreliable. A read is not something I was doing so I am not sure how it affects my original finding (i only found it when testing via the i2c-dev tools).

  * In addition, the Grove Barometer BMP180 (i2c=0x77) will break _all_ of the high addresses even when it itself not addresses, merely by being on the i2c bus. How this is possible I do not know, given that an errant i2c device should only be able to pull the data line down but never force it high. This same device works fine on the Arduino Uno.

3. I have found a way to rebuild the Edison image from scratch (it is not in the Intel documentation), and it does work (though "configure_edison --version" will not return a value). My plan is to enable KGDBoC and i2c debugging support to help trace the issue from the kernel side.

4. NOTE: the J-5 "lcd.js" library does not return i2c errors, which is why this issue is hidden.

5. NOTE: the I2c setup script in the Intel Arduino for Edison manually is _not_ sufficient to set up the i2c bus. I had to run J-5 to get i2c into a proper state. It is also good to remember that the i2c bus is not initialized on boot, but maybe should be.

rwaldron, 2015-07-21:
This is really interesting information—thanks for the follow up.

dreyna7399, 2015-08-02:
1. SOLUTION

It turns out that not only is the Seeedstudio "Grove_-_Barometer_Sensor_(BMP180)" is a slow speed I2C device (100Kh), it you put it on a full speed I2C like the Edison's (400Kh) it DISTORTS the signals, even when not addressed.

This explains the symptoms, where it breaks the bus on the Edison but not the slower Arduino Uno.

What is particularly annoying is that this speed restriction is not noted on the SeeedStudio Wiki, which led me down this garden path of failure modes.

   http://www.seeedstudio.com/wiki/Grove_-_Barometer_Sensor_%28BMP180%29
   
I totally need to add a digital oscilloscope to my arsenal since the distorted signals totally fooled my logic analyzer.

The moral of this defect then is to beware older I2C devices, and try the slow I2C bus speed when you get obscure errors.

2. I2C Clock Speed Control

FYI: here is how you slow down the I2C bus clock on the Edison.

(a) I2C Standard speed

  $ echo std > /sys/devices/pci0000:00/0000:00:09.1/i2c_dw_sysnode/mode

(b) I2C Fast speed (Edison default)

  $ echo fast > /sys/devices/pci0000:00/0000:00:09.1/i2c_dw_sysnode/mode
  
Here is my source link:

  https://communities.intel.com/thread/56148
  
  
3. Modifying Kernel Flags for the Edison image

FYI, I was able to make an Edison debug kernel image. However, as a warning I will mention that with its Yocto Project 1.7 I could not get config settings to work from a layer as documented - I had to force them onto the end of the "defconfig" file (the alternate but less pure documented workflow).

(a) My workaround:

  $ cat debug_i2c.cfg
  # CONFIG_SERIAL_KGDB_NMI is not set
  CONFIG_CONSOLE_POLL=y
  CONFIG_I2C_DEBUG_CORE=y
  CONFIG_I2C_DEBUG_ALGO=y
  CONFIG_I2C_DEBUG_BUS=y
  CONFIG_KGDB=y
  CONFIG_KGDB_SERIAL_CONSOLE=y
  # CONFIG_KGDB_TESTS is not set
  # CONFIG_KGDB_LOW_LEVEL_TRAP is not set
  # CONFIG_KGDB_KDB is not set
  $
  $ bitbake linux-yocto -c kernel_configme
  $
  $ cat debug_i2c.cfg >> /opt/dreyna/maker/edison/edison-src/meta-intel-edison/meta-intel-edison-bsp/recipes-kernel/linux/files/defconfig
  $
  $ bitbake linux-yocto
  $
  
(b) I do know that the layer was registered, by examining the kernel bitbake environment, so it was not that.

  $ bitbake linux-yocto -e | grep "^sccs="
  sccs="/opt/dreyna/maker/test_i2c_layer/recipes-kernel/linux/files/debug_i2c.cfg /opt/dreyna/maker/edison/edison-src/meta-intel-edison/meta-intel-edison-bsp/recipes-kernel/linux/files/defconfig"
  $
  
(c) I do not know it this was a YP-1.7 issue or how the Edison BSP was created.

I will investigate.

FYI, I also had a lot of problems with (just) the "bitbake -c menuconfig linux-yocto" command on my two Ubuntu hosts (14.04 and 15.04) which may be too new for YP-1.7. I had to go into the linux build directory and run "make menuconfig" explicitly.

closed 2015-09-03 by rwaldron