rwaldron/johnny-five #870: LM335 uses K by default

ajfisher, 2015-07-21:
I didn't realise there was a different on this but the difference between an LM335 and LM35 is that LM35 uses C and LM335 uses K.

As such you can't use the same controller for both (which I mistakenly believed for ages). As such we need a new controller for the LM335.

I have code so I'll do a PR and add to docs etc as well. Will come up with an option for nodebots day so we don't have to push it through before then.

rwaldron, 2015-07-21:
Sounds good, but remember that the controller must define a celsius conversion as the baseline value.

ajfisher, 2015-07-21:
Yep. I have all of that working in a patched version of temperature.js

rwaldron, 2015-07-22:
PR?

ajfisher, 2015-07-22:
Will do on Sunday. Need to write the examples, do a board layout etc. Got
it working with a custom controller for the moment so good for the session
on Sat but will do the PR properly.

rwaldron, 2015-07-23:
Ok, I understand. I'd love to get that controller landed for you in time for Nodebots Day, but I realize that would mean breaking our release freeze :(

dtex, 2015-07-23:
I'm sure you guys already considered this so apologies in advance...

To keep it simple for users this weekend, the controller could be published on npm (name spaced to avoid clutter) and then just required in.

ajfisher, 2015-07-23:
It's so little I just wrote it and put it in the repo for the kits we're
using.

ajfisher, 2015-07-26:
Okay - just got to write the tests and make a component for the fritzing diagram. Everything else is done and haven't had to write a J5 test before so that's going to be an experience ;)

closed 2015-09-03 by rwaldron