rwaldron/johnny-five #871: Add "documentation" property to each of the topics and examples in programs.json

dtex, 2015-07-21:
Would be a URL to relevant page(s)/anchor(s) at http://johnny-five.io/api/. Could be used to populate the readme and the md files in docs.

rwaldron, 2015-07-21:
This gets generated by johnny-five.io's build process, I bet we can just use the same process?

dtex, 2015-07-21:
Not sure I follow. I thought the readme and md files in docs were handled by J5's own grunt process.

rwaldron, 2015-07-21:
They are, but they don't match "component classes used in example X" to the appropriate api documentation. johnny-five.io's grunt tasks do more interesting things, which generate these: 

- http://johnny-five.io/examples/led/#component-classes-in-this-example
- http://johnny-five.io/examples/servo-animation-leg/#component-classes-in-this-example
- http://johnny-five.io/examples/grove-lcd-rgb-temperature-display/#component-classes-in-this-example

dtex, 2015-07-21:
Oh neat, I didn't know about that. My main goal was to funnel users from the docs folder to johnny-five.io/api. Maybe there is a better way to achieve that?

Resseguie, 2015-07-22:
@dtex this is what I had in mind in #705, but that was pre-website. Same idea.

rwaldron, 2015-11-05:
> My main goal was to funnel users from the docs folder to johnny-five.io/api

Let's definitely do this

closed 2018-08-28 by stale[bot]