rwaldron/johnny-five #873: Request - Button hold, continually trigger action

kevincannon, 2015-07-22:
I've a small feature request. In the Button object, there's a hold trigger. However, it only triggers after 500ms, and never again. Often press & hold is something where you need to repeatedly do the action like a keyboard. It woud be nice to have a repeat option built in. 

http://johnny-five.io/api/button/

Something like this:
```
rightButton   = new five.Button({
pin:3,
isPullup: true,
holdtime: 200,
holdrepeat: true
});

button.on("hold", function() {
    console.log("hold");
});
```
Then you would get this as an output
```
hold
hold
hold
```

It would be neat if it was built in!

rwaldron, 2015-07-22:
I like it and I think this might even be preferable as the default behavior? I won't be able to release b/c we're "frozen" until after Nodebots Day (this Saturday)—is that ok?

Resseguie, 2015-07-22:
I like it too.

kevincannon, 2015-07-24:
No rush at all, i've made a workaround for my needs for now, but built in would be even better :)

rwaldron, 2015-07-24:
@kevincannon thanks for patience!

closed 2015-07-24 by rwaldron

dtex, 2015-07-24:
Should this be a different event? Currently the hold event represents a state change. Also, we may want different behavior on that initial state change then we would want on the repeated firings of the event.

dtex, 2015-07-24:
Never mind, I didn't really think that through. Hold is not a state change.

rwaldron, 2015-07-24:
>  Also, we may want different behavior on that initial state change then we would want on the repeated firings of the event.

`button.once("hold", ...)`?

Resseguie, 2015-07-25:
I like that @rwaldron

kevincannon, 2015-07-27:
Wow guys - just checked that out, that's amazing. Thanks so much! :)

rwaldron, 2015-07-27:
Thank _you_ for filing.