rwaldron/johnny-five #874: Wifi connection?

Rickgg, 2015-07-23:
Hi, is there a way to connect an Arduino through wifi that can use Johnny-Five? Or what does IoT mean in this case?
Thanks a lot

rwaldron, 2015-07-23:
I haven't experimented with the Arduino WiFi shield, but only because the ones I've seen are pricey. Aside from that, there are other board options that use WiFi: Particle Core, Particle Photon and Electric Imp.

Resseguie, 2015-07-23:
Others, such as @ajfisher and @monteslu have done some work tunneling the serial communication via a WiFi connection. Here's one approach with cheap hardware:

https://gist.github.com/ajfisher/5fe60fe7d8c49b3223f0

ajfisher, 2015-07-23:
This gist is still 70% relevant. In light of events last night I'll update this to reflect the current approach we're working with for transparent bridges.

soundanalogous, 2015-07-25:
I can also create a WiFi version of StandardFirmata once I get ahold of a Wi-Fi shield. Currently most of the selection of WiFi shields for Arduino are too expensive so I've been avoiding it.

ajfisher, 2015-07-26:
That gist has now been updated with latest UDP transparent bridge - it's fast!

Rickgg, 2015-07-30:
Great!! Thanks!!

closed 2015-07-30 by Rickgg

rodrigoadachi, 2015-07-31:
talking about wifi and Arduino, it is less expensive the ENC28J60, however the library consumes much space. Or always say, we can develop pro ESP8266. I would like to help develop, how can I know the communication mode with the johnny-five via ethernet?