rwaldron/johnny-five #875: "ready" event not emitted by Boards

topkoa, 2015-07-23:
I have been attempting to connect multiple Arduinos to my project; however until now I have been unable to proceed by using the "Boards" API.

Without re-iterating; please see the following Stack Overflow question post on my issue (Update #3):

http://stackoverflow.com/questions/31495847/johnny-five-multiple-arduinos-connecting-but-not-emitting-ready

rwaldron, 2015-07-23:
I've found a bug in the repl initialization, this will be fixed in the next release.

topkoa, 2015-07-23:
Awesome! Thanks!
By the way-- the j5 api is absolutely amazing, you have done a fantastic job with it....  just wanted to let you know!

rwaldron, 2015-07-23:
> By the way-- the j5 api is absolutely amazing, you have done a fantastic job with it....  just wanted to let you know! 

<3

Thanks for the bug report!!

rwaldron, 2015-07-23:
Here's the full response from stack overflow: 

> I've found a bug in the repl initialization, this will be fixed in the next release. The issue was a call to `state.board.info(...)` in `Repl.prototype.initialize`, where the `info` method was undefined as a result of an oversight during recent refactoring of the board logging and logging function definitions. This bug appeared in v0.8.85, so try `npm install johnny-five@0.8.84` to resolve the issue until the next release.

closed 2015-07-23 by rwaldron