rwaldron/johnny-five #877: Play Nice with Node Debugger/Inspector

AdamHess, 2015-07-25:
Node debugger/nodeinspector makes the debugging experience the inspector crashes when any application is run please investigate.

rwaldron, 2015-07-28:
I won't have a chance to look at this until next week (I'm in seattle for tc39 meeting all week). Anyone else that wants to dig in should definitely do so

dtex, 2015-07-28:
@AdamHess I just gave it a shot with node-inspector 0.11.2, node 0.12.4, and Jonny-Five 0.8.85 and it all worked (inspected /eg/led.js). Were you using a different version of anything?

TheSpiderPig, 2015-07-28:
hmmm maybe i need to update my version, i ran it on Ubuntu if that changes anything.

dtex, 2015-07-28:
I didn't try Ubuntu (I was on OS X)  and that of course may change things. I'll have to get a VM set up so maybe someone else can take a look before I am able to get to that.

dtex, 2016-01-24:
Hi @AdamHess, did you ever have a chance to upgrade and try again?

AdamHess, 2016-01-24:
nope

closed 2016-08-26 by reconbot