rwaldron/johnny-five #882: Is there any way to use this with Docker ?

juliocesar-io, 2015-07-31:
It is possible to have serial communication with an Arduino on Raspberry pi 2 B+ (raspbian) using  Docker ?

Docker can access to attached devices using a volume

```
docker run --privileged -v /dev/bus/usb:/dev/bus/usb
```

And it works 

```
Bus 001 Device 005: ID 2341:0042 Arduino SA Mega 2560 R3 (CDC ACM)
```

But when I run my app It just remains looking for a connected device.

> Board Looking for connected device 

Any thoughts ?

node -v
v0.10.36

Resseguie, 2015-07-31:
I'll note that I'm potentially interested in the general Docker question as well, not just Raspbian.

rwaldron, 2015-08-03:
@nebrius @beriberikix @jfrazelle

rwaldron, 2015-08-03:
Did you try writing it out, like: 

```js
var five = require("johnny-five");
var board = new five.Board({ port: "/dev/ttyACM0" });

board.on("ready", function() {
   // ...
});
```

beriberikix, 2015-08-04:
It should definitely be possible. Do you know if the image has serial drivers installed? Seems like something one would strip out for the regular webapp use case.

juliocesar-io, 2015-08-10:
I set the port explicitly.

```js
var board = new five.Board({ port: "/dev/ttyACM0" });
``` 

And It just works like a charm.

Thanks for your help!!

closed 2015-08-10 by juliocesar-io