rwaldron/johnny-five #883: Multiple Arduino Boards on same USB Bus by Serial Number?

topkoa, 2015-08-03:
I'm curious if anyone has any insight into this question....    

A little background-- I'm working on a project that requires the ability to plug multiple Arduino boards into a powered USB hub, but if an Arduino gets unplugged and plugged back into another port the program needs to still wire the specific logic up to that specific board.. (ad hoc basically) so the logic will need to follow the boards by serial number I presume (which I was able to determine from the "pnpId" through the serialport module by the way.)

So far I have been marginally successful-- however I have hit a few roadblocks and wasn't sure if anyone had discussed this as a possible feature request?

Basically, in a nutshell, I would like to use the Five.Boards(), however I'd like to make sure a specific Arduino (and its connected pins) is always associated with its own "pnpId" serial number, or similar, no matter where the USB bus enumerates its location.

Ideally, what I would really love is to be able to do the following:

Original Code from the J5 API:
var ports = [
  { id: "A", port: "/dev/cu.usbmodem621" }, // <-- This changes if its plugged into another port between execution of the program for instance.
  { id: "B", port: "/dev/cu.usbmodem411" }
];

Note: I get this as the pnpId:  (gort scan serial)
"usb-Arduino__www.arduino.cc__0043_8543034323835151B141-if00"

Note: Below I use the following regex to just get the USB FTDI Serial number from that:
---> RegExp '.*_([^-]+)-[a-zA-Z0-9]+$'

What I would like to do (or similar):
var ports = [
  { id: "A", port: "8543632343135141C0A1" }, 
  { id: "B", port: "8543034323835151B141" }
];

...So when I reference the "board" that has an ID of "A" .. I would like be certain that in my database of devices the associated pin will correspond to the correct pins on that specific board.

I'd be happy to share some of the code I currently have in more detail, or if possible, tell me if I am barking up the wrong tree and need to have my head examined.

Thanks in advance!!

rwaldron, 2015-08-03:
I don't think this is supported by node-serialport.

Cc @jacobrosenthal

topkoa, 2015-08-03:
OK thanks, I guess I'll continue with my hacking away..  Right now I'm using serialport to get a list of ports/pnpId's and mapping them to boards after they are initialized.   I know its kindof a kludge...

rwaldron, 2015-08-04:
What are you using to connect the multiple boards to a single USB port? (So I can order one and maybe help this process)

topkoa, 2015-08-05:
I'm using this: "Sabrent 13 Port High Speed USB 2.0 Hub with Power Adapter And 2 Control Switches (HB-U14P)", (from Amazon).

However, as far as I know its not device specific-- I'm sure any powered USB hub will do.   

In my test environment I have 5 total Arduino's connected, (3 are UNO R3's  and 2 are Funduino MEGA's).  ... In case you are wondering I'm connecting them to a Raspberry PI Model 2 B+ as well...

As I stated above, my ultimate project I'm working toward needs to not care what port the Arduino board is connected to.  

I'd be happy to share more details if you are interested.

rwaldron, 2015-08-05:
> However, as far as I know its not device specific-- I'm sure any powered USB hub will do.

I've been conditioned to never assume this ;) 

I just order the hub and will check back when it arrives

dtex, 2016-01-24:
Jeff Hoefs has a trick that can help you with this:

> use a unique filename for the sketch loaded to each board. You can query the firmware name. It doesn't have to be named "StandardFirmata"

brownleeb, 2016-02-15:
Not sure how it works across various flavors of linux.  However, I am using Arch Linux on a model 2 Pi, and symbolic links are created in /dev/serial/by-id every time I plug in an arduino (I am currently using 2).  The name of the link appears to be from some of the attributes from udevadm info.  I cant seem to nail it down tho, appears to be concatenation of 'maufacturer', 'product',' idProduct'....  But its not consistent.
The Uno seems to use the serial number as part of the name.  I only have one Leonardo, so cant tell, but the Leo doesnt seem to have anything unique in its name, so there might just be an interface number assigned (possibly by the order it was initialized, which would break all the usages)...

reconbot, 2016-05-25:
@brownleeb the firmata filename trick is probably the best bet right now. Node Serialport has gone under some improvements around listing ports since this issue was opened but not every platform nor device supports all the meta fields on the port. 

I'm going to close this issue due to it's age, but if you'd like to continue with it feel free to reopen. Id also suggest opening the issue on either serialport or firmatajs. Once you can identify the ports via firmata you can give them an [ID](http://johnny-five.io/api/board/#parameters) in johnny-five.

closed 2016-05-25 by reconbot

CraigZych, 2018-02-01:
For anyone looking for a way to do this in the future I came up with a way I think works well. 
Forget the usb ids.

Dedicate 3-whatever pins as binary address pins and hook up to dip switches and if needed a couple extra to identify board type. Keep the pins consistent.
Then just initialize every detected board and read these pins to identify the board before continuing.