rwaldron/johnny-five #884: Add support for the HX711 Load Cell Amplifier

kthblmfld, 2015-08-08:
I am attempting to add a TAL220 10kg load cell to my robot with some assistance from the [HX711 load cell amplifier](https://www.sparkfun.com/products/13230). I have managed to get the latest driver installed on my Arduino and have found a few Javascript ports that I am trying to sort out. It would be nice if support for this were in johnny-five.

dtex, 2015-08-09:
@kthblmfld I did a load cell project last year. I didn't know there was a breakout board for this and had to wire up my own. Thanks for sharing. It's pretty easy to use this with Johnny-Five already. Just use an instance of [sensor](http://johnny-five.io/api/sensor/). 

This is an excerpt from my code. If it was empty, Twilio would call our receptionist with a gentle reminder (I have since been told that this was "not cool"). If it was recently filled it would tweet.

````
sensor = new five.Sensor({
    pin: "A1", // Or whatever "white" is hooked up to
    freq: 1000,
    threshold: 5
  });

sensor.on("data", function() {
    if(this.raw < 275 && status != 'empty') {
      status = 'empty';
      call();
    }
    if(this.raw > 460 && status != 'full') {
      status = 'full';
      tweet();
    }
  });
````

dtex, 2015-08-09:
We shouldn't close this issue until we add a load sensor example in /eg

kthblmfld, 2015-08-09:
@dtex Were you using the TAL220? It has a red(E+), black(E-), green(O+), and white(O-) wire coming off it. If so, what did you do with the green wire? Were you connecting it directly to the Arduino? Did you need to add any resistors?

Thanks for replying. I have been digging into this issue for a few weeks and departed from my comfortable realm of understanding some time ago  :)

dtex, 2015-08-09:
I was using an [INA125P](http://www.ti.com/lit/ds/symlink/ina125.pdf). It had the same wires and I switched out resistors until I found one that worked well for my weight range and load cell.

This [chart](http://www.controlweigh.com/loadcell_colors.htm) was invaluable for determining which wire did what.

dtex, 2015-08-09:
Sorry clicked the wrong button

kthblmfld, 2015-08-10:
I am able to successfully detect a change in load using the load sensor + HX711 and connecting to Arduino the Vcc, ground, and analog pins to 'DAT','CLK' running code in [these examples](https://github.com/sparkfun/HX711-Load-Cell-Amplifier). In this case, there is no need for additional resistors. 

Where I am having a hard time with johnn-five Sensor is the 'DAT' and 'CLK' pins. Sensor only takes one pin in the constructor. Which is it, and do I need to do something with the other one?

dtex, 2015-12-18:
@kthblmfld Whoah, I completely dropped the ball on this conversation. I know this is way too late but here it goes anyways...

It sounds like the HX711 uses SPI which firmata does not yet support. I found this proposal to add SPI to firmata here https://github.com/firmata/arduino/issues/245 Once the API is nailed down we can start writing SPI support into the various io-pugins and then Johnny-Five. 

If you want to use Johnny-Five, one option might be to use an I2C backpack. The process is explained here http://omit.io/2015/08/20/beginners-guide-to-backpacks/ It does require additional hardware and software, but would enable use of the HX711 with Johnny-Five.

I'm sorry I don't have better news, and I'm really, really sorry it took me months to see this again.

I'll leave this open because I'm sure SPI support will come. We can close it then.

kthblmfld, 2015-12-22:
> found this proposal to add SPI to firmata here firmata/arduino#245

Thanks Donovan,

No rush. I demoed the robot I was working on a few months back, and it has
been shelved while I go through a move. I do intend to pick it up again and
build out some features based on recommendations from the demo. It could be
a few months before I fire it up again  :)

Thanks!
Keith

shanek21, 2016-07-24:
Hey Donovan,

I am trying to stream data from a load cell to a website using johnny-five and pubnub. Any updates on support for the HX711 amplifier?

Thanks,
Shane

dtex, 2016-07-24:
No, sorry. 

SPI support in firmata is still somewhere off in the future https://github.com/firmata/arduino/issues/245

dtex, 2018-01-11:
Hi @kthblmfld  ,

Johnny-Five contributors and maintainers are loathe to close issues where someone has a need. We don't want anyone to ever feel that we don't care, but SPI support hasn't landed yet. Rather than leave it languishing as an open issue we have created a [Requested Features](../wiki/Requested-Features) page and added your request for the HX711 there.

closed 2018-01-11 by dtex

kthblmfld, 2018-01-11:
Works for me. Thanks for the heads up, Donovan.

-Keith

YosiLeibman, 2020-09-29:
Just to know, there was an update? in the docs I couldn't find it, what is the process of adding this type of things? maybe I can contribute it myself?

dtex, 2020-09-29:
That would be awesome! SPi support would open up a lot of things. Assuming you are using an Arduino compatible board running firmata, then this is the place to work on getting it added:

firmata/arduino#245

YosiLeibman, 2020-09-29:
I'm a pretty much newbie in the open-source world, so don't really know what it takes, its just to write a wrapper to some c library?