rwaldron/johnny-five #885: Button event not fired

kersten, 2015-08-12:
Hello,

sorry for opening an issue for this but I have no idea why my setup does not work.

I have an Intel Edison with an active low button circuit on gpio 14. The button works an the output of `cat /sys/class/gpio/gpio14/value` is 0 if the button is not pressed and 1 if the button is pressed. But the five.button won't call none of the callbacks.

Here is my code

```javascript
var five = require( 'johnny-five' ),
	Edison = require( 'edison-io' ),
	edison = new five.Board( {
		io: new Edison()
	} );

edison.on( 'ready', function () {
	var button = new five.Button( {
		pin: 'GP14',
		board: edison,
		holdtime: 1000,
		invert: false
	} );

	button.on( "down", function () {
		console.log( "down" );
	} );

	button.on( "hold", function () {
		console.log( "hold" );
	} );

	button.on( "up", function () {
		console.log( "up" );
	} );
});
```

Could you help me with that?

kersten, 2015-08-12:
This one works, and shows me 1 if i press the button. Did I miss anything in my first example?

```javascript
edison.digitalRead('GP14', function(data) {
        if (data != 0) {
            console.log(data);
        }
});
```

rwaldron, 2015-08-12:
Can you post a picture of your actual hardware—thanks!

rwaldron, 2015-08-12:
Unrelated to issue, but just a tip: you don't need to specify these: 

```
board: edison, // this is autodetected for you
invert: false, // this is the default 
```

:)

kersten, 2015-08-12:
[image]

rwaldron, 2015-08-12:
Cool thanks, I'll put together the same circuit and see what I can learn. Will post back as soon as I know more!

kersten, 2015-08-12:
Many thanks. I hope you find out what's going on ;)

rwaldron, 2015-08-12:
> I have an Intel Edison with an active low button circuit on gpio 14. The button works an the output of cat /sys/class/gpio/gpio14/value is 0 if the button is not pressed and 1 if the button is pressed

Are you sure? I'm seeing something very different. 

> This one works, and shows me 1 if i press the button. Did I miss anything in my first example?

That's correct, but it should return to 0 once you release; see below.

What I experienced is super strange and definitely not a software issue. I completely eliminated all "code" aspects and focused solely on the sysfs state w/r to the circuit itself. 

-----------

I assembled the button circuit: 

[image]

- SparkFun GPIO Block
- SPST momentary switch (aka "a button")
- 10kΩ resistor
- Jumpers:
  - Button signal -> GP14
    - signal -> multimeter + 
  - Button signal-resistor-ground -> GND
    - ground -> multimeter -
  - Button vcc -> 3.3V 
  

I've added comments to explain what I'm doing in real life:

```sh
# the button circuit is connected, but I haven't pressed anything yet. 
root@eddie:~# cat /sys/class/gpio/gpio14/value
0
# press and hold the button, while holding, ran this again
root@eddie:~# cat /sys/class/gpio/gpio14/value
1
# release the button and ran this again
root@eddie:~# cat /sys/class/gpio/gpio14/value
1
# Something is not right
root@eddie:~# cat /sys/class/gpio/gpio14/value
1
# The signal appears stuck?
root@eddie:~# cat /sys/class/gpio/gpio14/value
1
```

Here's two video demonstrations:

- SparkFun GPIO Block, unexpected behavior: 
  - https://www.youtube.com/watch?v=kG3a73zk0LI
- Intel Edison Arduino Board, expected behavior: 
  - https://www.youtube.com/watch?v=3mH2-sxHBf0

I'm going to reach out to friends at Intel and SparkFun to find out more.

kersten, 2015-08-12:
ok, very strange. Maybe this is what I saw. 

```bash
while true
do
    cat /sys/class/gpio/gpio36/value
    sleep 1
done
```

After pressing the button I saw 1, 1, 1, 0, 1, 0, 1 and so on. I then replaced the 10k with a 1k resistor and after that I had no problems anymore. Could you check that?

kersten, 2015-08-12:
Anyway, shouldn't the lib execute the callback?

rwaldron, 2015-08-12:
> I then replaced the 10k with a 1k resistor and after that I had no problems anymore.

I actually had to use a 100Ω to make it work. That needs to go into the docs.  

> Anyway, shouldn't the lib execute the callback?

When I posted the response above, I hadn't gotten this far—the other issue side tracked me. Since swapping out for 100Ω resistor, I've tested the same script with these: 

- Arduino Uno (reality check)
- [Edison + Arduino Board (w/ 10kΩ resistor)](https://www.youtube.com/watch?v=DSEcyTNLQq8)
- [Edison + Miniboard (with 10kΩ resistor)](https://www.youtube.com/watch?v=GGefO-Y0Has)
- [Edison + DFRobot IO Expansion (with 1kΩ resistor)](https://www.youtube.com/watch?v=Xxyq4xk1v-E)
- [Edison + SparkFun GPIO Block (with 100Ω resistor)](https://www.youtube.com/watch?v=kfzNDjWlUtM)

The links are to videos I took demonstrating the correct behavior, and I wasn't able to reproduce the original reported bug.

kersten, 2015-08-12:
Ok, then I will check it with a 100Ω resistor. Maybe this will solve my problem too.

kersten, 2015-08-13:
Didn't work for me. I had to use a 1kΩ resistor. But the Button won't call the callback :cry: I have the exact same circuit like you. The strange thing is, that `cat /sys/class/gpio/gpio14/value` shows me the correct value.

kersten, 2015-08-13:
Me again, It works with the 100Ω resistor, don't know what it was, but the callback doesn't work :cry:

ShawnHymel, 2015-08-13:
I wish I knew the exact reason why this is the case, but you do need to use a stronger (e.g. lower value) pull up with the GPIO block. 10k does not work, but I've had good luck using 2.2k (or lower) value resistors. The best I can figure is that either A) the TXB0108 level shifters have a weak pull up or B) there is a bug in the Edison that prevents you from disabling the internal weak (100k-ish?) pull ups on the GPIO. Both of these will require some more research on my part.

If you have the Mini Brearkout Board for the Edison, could you try measuring the voltage of a GPIO pin when you change the /gpioXX/direction to "in"? If it is pulled to 1.8V, you can assume there is an enabled internal pull up. If you put a 10k to ground from the pin and measure the pin again, if it is not 1.8V (somewhere in between), then you know there is a pull up. You can then take the voltage to figure out what the internal resistor is by solving a voltage divider equation.

rwaldron, 2015-08-13:
@kersten ok, let's collect information then :)

- Do `npm list --depth=0` and report the output
- Do `node -v` and report the output

kersten, 2015-08-13:
`npm list --depth=0`

```bash
klarashaus@0.0.1 /home/root/.node_app_slot
├── async@1.4.0 extraneous
├── edison-io@0.8.18
├── galileo-io@0.8.18
├── johnny-five@0.8.85
├── just.randomcolor@0.1.1
├── moment@2.10.6 extraneous
├── mqtt@1.3.5
├── node-hue-api@1.1.1 extraneous
├── node-pixel@0.4.1
├── onecolor@2.5.0
├── parseUri@1.2.3-2 extraneous
├── q@1.0.1 extraneous
├── request-util@0.1.0 extraneous
├── serialport@1.7.4 extraneous
├── stochator@0.4.0
├── traits@0.4.0 extraneous
└── xml2js@0.2.8 extraneous

npm ERR! extraneous: async@1.4.0 /home/root/.node_app_slot/node_modules/async
npm ERR! extraneous: serialport@1.7.4 /home/root/.node_app_slot/node_modules/serialport
npm ERR! extraneous: traits@0.4.0 /home/root/.node_app_slot/node_modules/traits
npm ERR! extraneous: parseUri@1.2.3-2 /home/root/.node_app_slot/node_modules/parseUri
npm ERR! extraneous: moment@2.10.6 /home/root/.node_app_slot/node_modules/moment
npm ERR! extraneous: node-hue-api@1.1.1 /home/root/.node_app_slot/node_modules/node-hue-api
npm ERR! extraneous: request-util@0.1.0 /home/root/.node_app_slot/node_modules/request-util
npm ERR! extraneous: xml2js@0.2.8 /home/root/.node_app_slot/node_modules/xml2js
npm ERR! extraneous: q@1.0.1 /home/root/.node_app_slot/node_modules/q
npm ERR! not ok code 0
```

node version: v0.10.38

rwaldron, 2015-08-14:
Are you using the Intel IoTDK thing? 

Please update to the latest version of Johnny-Five— i'm not sure if it'll make a difference but it will at least ensure that we're referring to the same code base.  Can you also try writing a totally different program, for example blinking an LED using the `Led` class.

kersten, 2015-08-14:
@rwaldron WOW, ok now it's working :smile: it was the version, after a update it runs now

rwaldron, 2015-08-14:
Haha, remember when I wrote this...

>  i'm not sure if it'll make a difference but it will at least ensure that we're referring to the same code base

Derp. 

> WOW, ok now it's working :smile: it was the version, after a update it runs now

This is the best news :) Please close the issue whenever you're ready—thanks!!

kersten, 2015-08-16:
OkOK, now I have one button and 2 switches, and everything works fine :+1:

closed 2015-08-16 by kersten