rwaldron/johnny-five #886: Lining One - Cannot call method 'includes' of undefined

pxlhero, 2015-08-15:
I'm running LininoOS, LininoIO with nino-io. It connects to the board fine and it works with this.pinMode and digitalWrite, but when I switch to new five.Led(), it gives me this issue:

```
TypeError: Cannot call method 'includes' of undefined
    at Pins.(anonymous function) [as isPwm] (/opt/project/node_modules/johnny-five/lib/board.pins.js:244:39)
    at Led.Controllers.DEFAULT.initialize.value (/opt/project/node_modules/johnny-five/lib/led/led.js:91:30)
    at new Led (/opt/project/node_modules/johnny-five/lib/led/led.js:209:10)
    at Board.<anonymous> (/opt/project/blink.js:9:11)
    at Board.EventEmitter.emit (events.js:92:17)
    at process._tickCallback (node.js:415:13)
```

When installing johnny-five, I had the compiling errors with serialport so I used this: http://wiki.linino.org/doku.php?id=wiki:nodejs to add it to /opt/usr/lib/node_modules. Not sure if that has anything to do with it, or if I should install node-gyp and run it.

rwaldron, 2015-08-16:
Thanks for the report! Can you paste the code you're running?

pxlhero, 2015-08-16:
Here it is, pretty straightforward.

```
var five = require("johnny-five");
var Nino = require("nino-io");

var board = new five.Board({
      io: new Nino()
});

board.on("ready", function() {
     console.log("Ready.");
     var state = 0;
     var pin = 13;

    // this breaks
     var led = new five.Led(13);

     this.pinMode(pin, this.MODES.OUTPUT);
     
     setInterval(function() {
         board.digitalWrite(pin, (state ^= 1));
     }, 500);
});
```

rwaldron, 2015-08-17:
Probably unrelated, but you can't/shouldn't reuse pins like that. What happens when you just call `led.blink(500)`, instead of setting pinMode and using an interval?

pxlhero, 2015-08-17:
If I just delete everything in ready and just keep ```var led = new five.Led(13)``` I get the error. 
I did some investigation and the blink example in node_modules/firmata doesn't work either. ready doesn't fire at all, isReady is always false. 

If I install the FirmataSPI.hex from http://wiki.linino.org/doku.php?id=wiki:nodejs, firmata blink example works but when I run my code, I get the Failed to export PWN error, because /dev/ic2-0/ gets deleted and also get the error described here. Hope that helps.

rwaldron, 2015-08-17:
> Here it is, pretty straightforward.
> ...code...

You're running that code _on_ the Linino One, correct? And that code is running in the correct directory, right? There is a note about this in https://github.com/rwaldron/nino-io/#lininoos

>> IMPORTANT: The Linino expects all user code to exist in /opt! When connecting to the board, remember to: cd /opt. 

Sorry if that seems like a silly question, I just need to get a better understanding. The error you're seeing looks like it might be the result of the "layout" not being loaded. 

> If I install the FirmataSPI.hex from http://wiki.linino.org/doku.php?id=wiki:nodejs, firmata blink example works but when I run my code, I get the Failed to export PWN error, because /dev/ic2-0/ gets deleted and also get the error described here. Hope that helps.

Well, don't do that :P Nino-IO doesn't rely on that or any other "special firmata" thing. In fact, it doesn't use firmata at all on this platform. All you need is LininoOS & LininoIO.

pxlhero, 2015-08-17:
Yep, I'm running it on the Linino One in the opt directory. I'm going to reset the board and do it all over again. Should I just ignore the serialport errors I get from npm install johnny-five? Or should I still use the one provided by linino?

rwaldron, 2015-08-17:
> Should I just ignore the serialport errors I get from npm install johnny-five?

Yes—serialport is an optional dependency. 

> Or should I still use the one provided by linino?

No serialports are used by Johnny-Five and Nino-IO on Linino One

pxlhero, 2015-08-18:
I'm getting the same error after I reset the board. These were my steps:
- Install/confirm I have the latest LininoOS and LininoIO. I have the /dev/ic2-0 folder
- Install node, the /opt gets created
- Download and run serialTerminal.hex
- Install johnny-five and nino-io in /opt
- Create blink.js in /opt, add ```var led = new five.Led(13)``` on board.ready.
- run node blink.js

rwaldron, 2015-08-18:
Ok, thanks for running through it again. I will try to repro today.

rwaldron, 2015-08-21:
Of course, now I'm having trouble with this step: 

> Install node, the /opt gets created

I will keep you posted. Sorry about this taking so long :|

rwaldron, 2015-08-21:
Ok, got through that issue. I think I found the bug as well. I will let you know if that's true once I confirm

pxlhero, 2015-08-21:
Awesome, thanks a lot!

rwaldron, 2015-08-21:
Can you try nino-io@0.2.1? Thanks!

pxlhero, 2015-08-22:
Confirming it works now. Thanks!

closed 2015-08-22 by pxlhero

rwaldron, 2015-08-24:
:clap: