rwaldron/johnny-five #887: GY-IICLCD (PCF8574T) LCD Backpack On 20x4 LCD

Zenitram-Oriaj, 2015-08-15:
Been having an issue showing any text on my LCD. 

Whenever I start/restart my node app, the LCD cursor moves a couple positions but that is it.

I have tested it with a basic Arduino program on a Pro Mini

[image]

But when I run my program, this is all I get:
[image]

Here is the "lcd" object I created:

    lcd = new five.LCD({
        controller: "PCF8574T",
        address: 0x20,
        rows: 4,
        cols: 20
    });

Here is inside my current project:
[image]

Hardware:
Raspberry Pi B+
Arduino Pro Mini (Clone) with "Standard Firmata"

Node.JS v0.10.36
Johnny-Five v0.8.86

I have tried a number of ideas, and research all through this GIT repository, but have ran out. 
Please help.

rwaldron, 2015-08-16:
Thanks for the detailed report! Can you also paste the rest of the code you're running?

Zenitram-Oriaj, 2015-08-17:
Per your request. I hope it is sufficient.  Thanks for any help you can provide.
```javascript
/**
 * Created by Jairo Martinez on 8/15/15.
 */

var five = require("johnny-five");

var board = {};
var lcd = {};
var rly1 = {};
var rly2 = {};

var btnBack = {};
var btnMenu = {};
var btnHome = {};
var btnEntr = {};
var btnNvUp = {};
var btnNvDn = {};
var btnNvLt = {};
var btnNvRt = {};

function _btnsSetup() {
	btnEntr = new five.Button({
		pin:      2,
		isPullup: true
	});

	btnHome = new five.Button({
		pin:      3,
		isPullup: true
	});

	btnBack = new five.Button({
		pin:      8,
		isPullup: true
	});

	btnMenu = new five.Button({
		pin:      9,
		isPullup: true
	});

	btnNvRt = new five.Button({
		pin:      5, //
		isPullup: true
	});

	btnNvDn = new five.Button({
		pin:      4, //
		isPullup: true
	});

	btnNvUp = new five.Button({
		pin:      7, //
		isPullup: true
	});

	btnNvLt = new five.Button({
		pin:      6,
		isPullup: true
	});
}

function _btns() {
	_btnsSetup();

	btnMenu.on("down", function () {
		console.log("Btn Menu");
		lcd.print("Menu");
	});

	btnHome.on("down", function () {
		console.log("Btn Home");
	});

	btnEntr.on("down", function () {
		console.log("Btn Enter");
	});

	btnBack.on("down", function () {
		console.log("Btn Back");
	});

	btnNvUp.on("down", function () {
		console.log("Btn Up");
	});

	btnNvDn.on("down", function () {
		console.log("Btn Down");
	});

	btnNvLt.on("down", function () {
		console.log("Btn Left");
	});

	btnNvRt.on("down", function () {
		console.log("Btn Right");
	});
}

function _lcd() {
	lcd = new five.LCD({
		controller: "PCF8574",
		address:    0x20,
		rows:       4,
		cols:       20
	});

	lcd.clear();
	lcd.home();
}

/**
 * Setup Relays
 * @private
 */
function _relays() {
	rly1 = new five.Relay(10);
	rly2 = new five.Relay(11);

	setTimeout(function () {
		rly1.on();
		rly2.on();
	}, 1000);
}

/**
 * Initialize
 * @private
 */
function _init() {
	_btns();
	_lcd();
	_relays();
}

///////////////////////////////////////////////////
//

var j5 = function (){

};

/**
 * Initialize Module
 * @param opts
 * @param cb
 */
j5.prototype.init = function (opts, cb) {
	cb = (typeof cb === 'function') ? cb : function () {};

	opts = opts || { port: "/dev/ttyAMA0" };
	board = new five.Board(opts);

	try {
		board.on("ready", function () {
			_init();

			this.repl.inject({
				lcd: lcd
			});

			cb(null);
		});
	}
	catch(ex){
		cb(ex,null);
	}
};

/**
 * Send Message To The LCD
 * @param msg
 * @param row
 * @param col
 */
j5.prototype.message = function(msg,row,col){
	row = row || 0;
	col = col || 0;

	lcd.cursor(row, col).print(msg);
};

module.exports = new j5();
```

Zenitram-Oriaj, 2015-08-21:
Any updates on this issue?

rwaldron, 2015-08-22:
I'm so sorry, I completely missed your response with the example code. I'm away for the weekend but I will try to get this reproduced on Monday. I really appreciate your patience and I apologize for the delay.

Zenitram-Oriaj, 2015-08-23:
Thanks for letting me know your out this weekend.

In the mean time, I have made a little progress on this issue. Using the manufacture provided C header file, I did see some differences between OPTS Defaults in the /lib/lcd.js file and in the #define of the LiquidCrystal_I2C.h file:

```c
// flags for backlight control
#define LCD_BACKLIGHT 0x00
#define LCD_NOBACKLIGHT 0x80

#define En B00010000  // Enable bit
#define Rw B00100000  // Read/Write bit
#define Rs B01000000  // Register select bit 
```
Hex values for EN, RW, RS are 0x10, 0x20, 0x40

here is a link where you can download the C Library for this LCD Backpack
http://artofcircuits.com/product/i2c-io-expander-interface-board-16x2-20x2-character-lcd-modules

I did another series of test using some different hardware as shown in the attached images:
[image]

Same LCD Backpack but on a LCD 16x2 display.
Same Arduino Pro Mini running "Standard Firmata".
In this test, I am using my Radxa Rock running Linaro 14.04 / Node v0.12.3

In the /lib/lcd.js files, I made these edits:
(Assuming that the first 4 bits are the outputs connected to D4-D7)
```js
var OPS = {
  DEFAULT: {
  ...
    // Enable
    EN: 0x10,
    // Read/Write
    RW: 0x20,
    // Register Select
    RS: 0x40,

    // DATA
    D4: 0x01,
    D5: 0x02,
    D6: 0x04,
    D7: 0x08
  }
}
```
As you can see in the attached image below, I am now able to get text to show on the LCD but NOT the text I am trying to display.
[image]

PLEASE NOTE: I am by no means an expert in I2C (TWI) communications in any sense. I am still trying to understand how it works. So the assumptions I have made maybe way off course with how your library actually works.

rwaldron, 2015-08-24:
> `controller: "PCF8574",`

Can you change this to `PCF8574T` and try you code? I just tried this: 

```js
var five = require("../lib/johnny-five");
var board = new five.Board();

board.on("ready", function() {
  var l = new five.LCD({
    controller: "PCF8574T",
    rows: 4,
    cols: 20
  });

  l.useChar("heart");
  l.cursor(0, 0).print("hello :heart:");
  l.blink();
  l.cursor(1, 0).print("Blinking? ");
  l.cursor(2, 0).print("3rd row");
  l.cursor(3, 0).print("4th row");
});
```

And it produced this: 

[image]

dtex, 2016-01-24:
Hi @Zenitram-Oriaj, any updates on this issue?

dtex, 2018-01-16:
@Zenitram-Oriaj I'm going to close this since we haven't heard back in a while. Feel free to re-open if you have any updates.

closed 2018-01-16 by dtex