rwaldron/johnny-five #890: iojs v3.2.0 can't complete tests.

henricavalcante, 2015-08-25:
````
[20:38:51] henricavalcante@machine:~/git/johnny-five (master)
$ grunt nodeunit:complete
Running "nodeunit:complete" task

Running "nodeunit:tests" (nodeunit) task
Testing accelerometer.js................OK
Testing animation.js..........OK
Testing barometer.js....OK
Testing board-connection.js.OK
Testing board.component.js.........................OK
Testing board.js..................................OK
Testing board.pins.js.......OK
Testing button.js............OK
Testing capabilities.js.......OK
Testing collections.js....OK
Testing compass.js..........OK
Testing esc.js......................OK
Testing expander.js...................................................................OK
Testing fn.js.........OK
Testing gyro.js.......OK
Testing imu.js......OK
Testing joystick.js..............OK
Testing keypad.js...............................OK
Testing lcd.js......................OK
Testing led.js................................................OK
Testing ledcontrol.js.......................................................OK
Testing motion.js..............................OK
Testing motor.js...........................................................OK
Testing options.js.OK
Testing piezo.js...Error: Please update to the latest version of node!
F
>> Incomplete tests/setups/teardowns:
>> Piezo - tone 0 [ 'Piezo - tone' ]
Fatal error: A test was missing test.done(), so nodeunit exploded. Sorry!

[20:39:13] henricavalcante@machine:~/git/johnny-five (master)
$ node -v
v3.2.0

````

rwaldron, 2015-08-26:
Thanks for the report!!

closed 2015-08-26 by rwaldron

rwaldron, 2015-08-26:
np—Thanks again for the report!