rwaldron/johnny-five #893: Can't control servo in latest raspberry pi 2

born2net, 2015-08-26:
I have 2 lines of code:

var five = require("johnny-five");
var board = new five.Board();

and get an error of:

```
/root/node_modules/johnny-five/lib/board.js:83
      ports = result.filter(function(val) {
                    ^
TypeError: Cannot read property 'filter' of undefined
    at Board.<anonymous> (/root/node_modules/johnny-five/lib/board.js:83:21)
    at /root/node_modules/johnny-five/node_modules/serialport/serialport.js:565:11
    at FSReqWrap.oncomplete (fs.js:95:15)
```

I can control LEDs just fine with johnny five. 
Are servos not supported?

regards

rwaldron, 2015-08-26:
Hi, thanks for the report. Are you trying to control servos connected directly to the raspberry pi? If so, then you'll need to use the appropriate IO Plugin: https://github.com/nebrius/raspi-io

```js
var raspi = require('raspi-io');
var five = require('johnny-five');
var board = new five.Board({
  io: new raspi()
});

board.on('ready', function() {
  //...
});
```

born2net, 2015-08-26:
thank you... trying now...

born2net, 2015-08-26:
THANK YOU rwaldron

worked great

```

var raspi = require('raspi-io');
var five = require('johnny-five');
var board = new five.Board({
  io: new raspi()
});

board.on("ready", function() {
  var servo = new five.Servo("P1-12");
   servo.sweep();
});

```

closed 2015-08-26 by born2net

rwaldron, 2015-08-26:
My pleasure—glad we could get you sorted out so quickly :D