rwaldron/johnny-five #895: WIFI Connect - Shield WIFI ESP8266

rudemex, 2015-08-27:
Hi, how to connect wifi with shield WIFI ESP8266?

i testing this source https://gist.github.com/ajfisher/5fe60fe7d8c49b3223f0 , but no work :/

Thanks

monteslu, 2015-08-27:
You'll need to test to make sure the esp image is working.  Can you connect to it's wifi access point from your computer?

rudemex, 2015-08-27:
no, from the computer I can not connect to the shield

monteslu, 2015-08-27:
That's the hard part :)

Only advice I have on that is that I had better luck using esptool in Linux than Mac, and to make sure you're using 3.3v for power as well as RX/TX.  If the firmware writes correctly it will have an open access point available when it restarts.

rudemex, 2015-08-27:
you could run on mac?

monteslu, 2015-08-27:
Yeah it definitely runs on mac, just failed for me a couple of times while it worked on linux flawlessly.  Could have just been my system.

rudemex, 2015-08-27:
great, I have exactly the same connection,The only thing different is that I give energy to the usb connected to the Mac.

what steps it work?

Thanks

reconbot, 2016-05-25:
@rudemex if you're still interested I suggest reaching out to @monteslu on our [gitter channel](https://gitter.im/rwaldron/johnny-five?utm_source=badge&utm_medium=badge&utm_campaign=pr-badge&utm_content=badge). I'm going to close this for now as while it can be made to work, the esp8266 isn't something johnny-five actively supports at this time.

closed 2016-05-25 by reconbot

soundanalogous, 2016-05-25:
ESP8266 is supported actually but not well documented. You will need to use [StandardFirmataWiFi from the master branch of firmata/arduino](https://github.com/firmata/arduino/tree/master/examples/StandardFirmataWiFi) and see the wifiConfig.h file for ESP8266-related instructions and either [etherport](https://github.com/rwaldron/etherport) or [etherport-client](https://github.com/mwittig/etherport-client) depending on whether you configure the ESP8266 as a TCP/IP client or server.

See this thread for examples: https://github.com/firmata/arduino/issues/257.

frxnz, 2016-05-26:
@samjulien also wrote a nice blog post about getting set up with ESP8266 (specifically a Feather HUZZAH) and J5 that covers a lot of this http://www.samjulien.com/johnny-feather/