rwaldron/johnny-five #896: Proximity - HC-SR04 issue

carloscarcamo, 2015-08-28:
I'm getting this warning:

```sh
$ node app.js
1440742786705 Device(s) /dev/ttyACM0  
1440742786716 Connected /dev/ttyACM0  
1440742790038 Repl Initialized  
>> distance 4356.718
distance 33.505
distance 34.588
distance 4358.746
distance 4355.12
distance 4356.701
distance 4357.216
distance 4354.639
distance 0
distance 4356.22
distance 4354.931
distance 0
(node) warning: possible EventEmitter memory leak detected. 11 ping-read-7 listeners added. Use emitter.setMaxListeners() to increase limit.
Trace
    at addListener (events.js:179:15)
    at once (events.js:204:8)
    at Board.pingRead (/home/carloscarcamo/workspace/node/NodeBots/tests/node_modules/johnny-five/node_modules/firmata/lib/firmata.js:1334:8)
    at Proximity.<anonymous> (/home/carloscarcamo/workspace/node/NodeBots/tests/node_modules/johnny-five/lib/proximity.js:175:19)
    at wrapper [as _onTimeout] (timers.js:265:14)
    at Timer.listOnTimeout (timers.js:110:15)
distance 4356.151
distance 4355.137
distance 4355.653
```
I don't know if i'm doing something wrong, i'm using this simple code:

```js
var five = require('johnny-five'),
  board = new five.Board(),
  proximity = undefined;

board.on('ready', function(){  
  proximity = new five.Proximity({
    controller: 'HCSR04',
    pin: 7
  });

  proximity.on('change', function(){
    console.log('distance', this.cm);    
  });
});
```

Is this normal or there is any problem?

Thanks in advance!

topkoa, 2015-08-28:
I've had this issue before. 

If it helps at all I did not experience this with the more expensive "Parallax Ping Ultrasonic Range Sensor 28015", I ordered the "WYPH Ultrasonic Module HC-SR04" and had the issue that is mentioned above...  

I've got a whole mess of these things I'd like to use so I'm quite interested if there is any resolution to this as well!

rwaldron, 2015-08-28:
Did you load the correct firmware onto the board? These sensor require the custom [PingFirmata provided in the docs](http://johnny-five.io/api/proximity/#pingfirmata)

topkoa, 2015-08-28:
In my case; I tried re-uploading that firmata it several times.  I switched back and forth between the two models of ping sensors as well.     I haven't had it on the test bench in a few weeks but I can fire it back up and try is again this weekend.

rwaldron, 2015-08-28:
I have several models of this same sensor, which were used to test and confirm support and all of them work correctly. 

@carloscarcamo @topkoa can you both post pictures of your actual hardware? Thanks!

topkoa, 2015-08-28:
I sure will.   I won't have an opportunity until tomorrow afternoon though.

rwaldron, 2015-08-28:
Thanks! Be sure to @-ping me when you post :)

carloscarcamo, 2015-08-28:
@rwaldron yes, I've loaded the PingFirmata.
I'm using this model:

http://www.icstation.com/arduino-ultrasonic-module-sr04-distance-transducer-sensor-p-1389.html

Maybe the quality of the sensor is what cause the warning, maybe I need to buy a better sensor.

topkoa, 2015-08-28:
I think I have the same model (looks identical, same pins).   

I did buy that model for a couple reasons:

1) J5 is documented to support it, and:
2) Super cheap! (as opposed to the Parallax version (~$30 bux!)

I had the same feeling as you.. Maybe to cheap to be true? I guess we will see though. 

I have a feeling Rick will know what's up. :)

rwaldron, 2015-08-28:
Have you both included a jumper that connects the Echo and Trigger pins?

topkoa, 2015-08-28:
I'll have to check my test rig; however I believe it was indeed jumped.

carloscarcamo, 2015-08-28:
yes! I did it!, as this diagram suggests:

http://johnny-five.io/api/proximity/#hcsr04parallax-ping-

topkoa, 2015-08-28:
Nice! -- I'll still give it a check on my bench too when I get home tomorrow.

topkoa, 2015-08-28:
Welp; I got home early.  Just tried it and am getting the same problem...

I have it wired exactly as the diagram indicates and with the correct PingFirmata installed.  

I have both ping sensors hooked up as well; (both the Parallax and the Generic).  

Here's a PIC:
[image]

When I run the program, the generic ping sensor works for a few seconds then I get the same error as above.  When I change the pin to use the Parallax ping it works flawlessly.

Let me know if there is anything I can try.

topkoa, 2015-08-28:
Here is the wiring for both:

[image]
[image]

topkoa, 2015-08-28:
Another update:  I swapped out the Generic ping sensor with another with the same results.  Ruling out a faulty sensor.  (I got a pack of 10 of them)..  I only have one of the Parallax ping sensors, but that one seems to work just fine as stated above.

rwaldron, 2015-08-29:
Thanks for the follow up. Historically, the error you're seeing has been the result of insufficient power flow to the connected component. Let's try to reduce this as much as possible. 

- Remove everything from your Arduino 
- Connect _only_ the 4 pin HCSR04 (as shown in that diagram)
- **Do not** jump the 5V and GND from breadboard rails, plug those in directly to 5V and GND

[image]

If this doesn't correct the problem, then please take another picture of the _complete_ circuit, which should show: 

- Arduino
  - Power jumpers
  - Signal jumper
- Breadboard + Ping sensor

rwaldron, 2015-08-29:
Also, this is an interesting read: https://forum.sparkfun.com/viewtopic.php?f=42&t=40513

carloscarcamo, 2015-08-29:
Ok. I will try later and post my results.
I think this time will work correctly :)
Yesterday I tested the ultrasonic sensor with pure C code with the arduino IDE and it worked perfectly, this was the diagram: 

[image]

And the C code:

https://gist.github.com/carloscarcamo/2d2bf6a22ef8c7f081a1

rwaldron, 2015-08-29:
@carloscarcamo can you try exactly as I describe here https://github.com/rwaldron/johnny-five/issues/896#issuecomment-136003019 Thanks!

carloscarcamo, 2015-08-29:
Sure, I will do it when I get home in a couple of hours :)

topkoa, 2015-08-29:
So I did what you said, wired exactly as you indicated.  I also used a multimeter and the voltage us a steady 4.92. 

I'll post some pics soon for you to verify.  

From reading that thread it looks like they are just garbage.  Hah.  

Maybe put a warning in the wiki? 

Want me to mail you one so you can see for yourself?

rwaldron, 2015-08-29:
> Maybe put a warning in the wiki?

Probably a good idea. 

> Want me to mail you one so you can see for yourself?

I have soooo many of these—but all of mine work correctly. Where are you located?

topkoa, 2015-08-29:
Coventry, Rhode Island

rwaldron, 2015-08-29:
Oh, that's not bad. Send me an email to waldron.rick@gmail.com and we can arrange this

carloscarcamo, 2015-08-29:
The same issue!
I think that the quality of the sensor is what cause the problem! 

[image]

rwaldron, 2015-08-29:
Duuuuude, is there anything between the bottom of your arduino and that _metal_ table??

topkoa, 2015-08-29:
[image]

Here's my test; and mine is on a static sheet!  (sorry! had to digg!)

carloscarcamo, 2015-08-30:
it seems like a metal table but it is not, it's hard to explain jajaja!

rwaldron, 2015-09-02:
@topkoa I received the two devices today and just set up some tests, compared them with several of my own and I'll be damned if I can't figure out what's going on with these. I can modify the read handler loop so that it only requests another measurement after the last measurement has been received. There are other controllers that do this, so it's not an issue. This will guarantee that it's impossible for another event handler to be added before the last one has cleared—this will fix the error that we're all now seeing.

rwaldron, 2015-09-02:
Just to make sure it's documented: this fix is working around faulty hardware that's not implementing the trigger/echo timeout correctly.

closed 2015-09-02 by rwaldron

topkoa, 2015-09-02:
Awesome; glad I could help!

topkoa, 2015-09-02:
Just confirming the fix worked!  I'm no longer getting that error on your latest code.

A side note, these defective sensors are still effectively not good for a production application in my opinion-- with it running on my bench it occasionally just stops reading distance (returns 0) and sits idle for a good 30 seconds then seems to reset. I presume this is the condition where it caused J5 to crash before the fix.

rwaldron, 2015-09-03:
>  with it running on my bench it occasionally just stops reading distance (returns 0) and sits idle for a good 30 seconds then seems to reset. I presume this is the condition where it caused J5 to crash before the fix. I presume this is the condition where it caused J5 to crash before the fix.

Yep, thats what I'm referring here: 

>> this fix is working around faulty hardware that's not implementing the trigger/echo timeout correctly.

Basically, we're just being more graceful about how the flaw is handled.

carloscarcamo, 2015-09-04:
@rwaldron thanks for taking some time and fix the problem :)

rwaldron, 2015-09-04:
No problem!