rwaldron/johnny-five #898: Can't get Raspberry pi working LCD 16x2

born2net, 2015-08-30:
Hello

here is the code

```
#!/usr/local/bin/node
var five = require("johnny-five");
var raspi = require('raspi-io');
var board = new five.Board({
  io: new raspi()
});

board.on("ready", function() {
    io: new raspi()

  var lcd = new five.LCD({
    // LCD pin name  RS  EN  DB4 DB5 DB6 DB7
    // Arduino pin # 7    8   9   10  11  12
    // pins: [37, 35, 33, 31, 29, 23],
    pins: ["GPIO26", "GPIO19", "GPIO13", "GPIO6", "GPIO5", "GPIO11"],
    backlight: 10,
    rows: 2,
    cols: 16
  });

  var frame = 1;
  var col = 0;
  var row = 0;

  lcd.display();
  lcd.useChar("runninga");
  lcd.useChar("runningb");

  this.loop(300, function() {

    lcd.clear().cursor(row, col).print(
      ":running" + (++frame % 2 === 0 ? "a" : "b") + ":"
    );

    if (++col === lcd.cols) {
      col = 0;

      if (++row === lcd.rows) {
        row = 0;
      }
    }
  });
});
```

I tried both pins and GPIO and I get a crash with:

```
1440904481393 Device(s) RaspberryPi-IO
1440904481434 Connected RaspberryPi-IO
1440904481456 Repl Initialized
>> /root/johnyfive/lcd_16x2.js:25
  lcd.display();
      ^
TypeError: undefined is not a function
    at Board.<anonymous> (/root/johnyfive/lcd_16x2.js:25:7)
    at Board.emit (events.js:104:17)
    at process._tickDomainCallback (node.js:381:11)
PI-~/johnyfive->
PI-~/johnyfive->
PI-~/johnyfive->
PI-~/johnyfive-
```

I know hardware with everything is working since it works via Python...

any help is appreciated

regards

rwaldron, 2015-08-30:
That second `io: new raspi()`, inside the board's ready handler is probably causing trouble

born2net, 2015-08-30:
tried it without didn't work.. ended up using this lib: https://github.com/fivdi/lcd
worked great with that lcd lib

closing issue...

tx for comment

closed 2015-08-30 by born2net

rwaldron, 2015-08-31:
They do the same thing 0_o

The problem was that the program called `display()` instead of `on()`, which is my fault since it was never update in the docs, nor in the runner example. I've just updated both.

born2net, 2015-08-31:
tx!!