rwaldron/johnny-five #899: Led pulse consuming a lot of CPU

gfpacheco, 2015-08-31:
Just by running the [Led - Pulse](https://github.com/rwaldron/johnny-five/blob/master/docs/led-pulse.md) sample code I had my CPU drained by Node. While changing `led.pulse()` for `led.blink()` reduced the usage almost entirely.

dtex, 2015-08-31:
Yep, I think we've got a problem with code that uses Animation() and loops or runs indefinitely like pulse(). The question is how best to handle it. Our current solution is cool in that it gives us sub ms timing but we don't always need that.

@rwaldron, spit balling here: what do you think of Animation() using temporal at first but setting thresholds on duration? If those thresholds are exceeded we drop back to setInterval and a reasonable  refresh rate (which can be inferred based on the frame rate when running in those first few seconds under temporal).

rwaldron, 2015-09-01:
@dtex 

> what do you think of Animation() using temporal at first but setting thresholds on duration? If those thresholds are exceeded we drop back to setInterval and a reasonable refresh rate (which can be inferred based on the frame rate when running in those first few seconds under temporal).

This is exactly what I'm thinking. Go for it. 

Also, yesterday I updated https://github.com/rwaldron/temporal to include explanations about it's use and why the CPU usage spikes.

dtex, 2015-11-06:
@rwaldron Okay, I need to get on this before we melt someone's CPU. What do you think of putting the threshold mechanism into Temporal instead of Animation()?

rwaldron, 2015-11-06:
@dtex not sure—can we try Animation first and see if it makes sense to move to Temporal?

dtex, 2015-11-06:
Will do.

dtex, 2015-12-16:
Added a fix in #982. When that lands I will close this.

dtex, 2015-12-16:
Landed

closed 2015-12-16 by dtex