rwaldron/johnny-five #900: OneWire thermometer device addresses disappearing

talexand, 2015-09-04:
## Equipment

- Board: Arduino Uno (clone) with example ConfigurableFirmata sketch burned in
- Hardware: MAX31850(2) with bidirectional 3.3-5V level shifter
- Software: johnny-five at  ^0.8.89
  - Modification: `console.log("Devices: ", devices);` inserted on line 59

Will post wiring photo shortly.

## Expected behavior:

Devices report temperature and address

## Log of actual result

Results are identical excepting `devices.length` if one board is removed. Arduino OneWire Tester sketch polls devices as wired properly.

```
➤ node temperature.js
1441398157817 Device(s) /dev/ttyACM0  
1441398157833 Connected /dev/ttyACM0  
1441398161293 Repl Initialized  
>> devices:  [ [ 59, 138, 118, 24, 0, 0, 0, 82 ],
  [ 59, 241, 114, 24, 0, 0, 0, 45 ] ]
/home/sobakasu/unisync/projects/autonano/node_modules/johnny-five/lib/temperature.js:39
            result = result * 256 + device[i];
                                          ^
TypeError: Cannot read property '6' of undefined
    at getAddress (/home/sobakasu/unisync/projects/autonano/node_modules/johnny-five/lib/temperature.js:39:43)
    at EventEmitter.<anonymous> (/home/sobakasu/unisync/projects/autonano/node_modules/johnny-five/lib/temperature.js:103:35)
    at null.<anonymous> (/home/sobakasu/unisync/projects/autonano/node_modules/johnny-five/node_modules/firmata/lib/firmata.js:1206:5)
    at g (events.js:199:16)
    at emit (events.js:107:17)
    at Object.SYSEX_RESPONSE.(anonymous function) (/home/sobakasu/unisync/projects/autonano/node_modules/johnny-five/node_modules/firmata/lib/firmata.js:313:9)
    at Object.SYSEX_RESPONSE.(anonymous function) (/home/sobakasu/unisync/projects/autonano/node_modules/johnny-five/node_modules/firmata/lib/firmata.js:291:29)
    at null.<anonymous> (/home/sobakasu/unisync/projects/autonano/node_modules/johnny-five/node_modules/firmata/lib/firmata.js:483:48)
    at SerialPort.emit (events.js:107:17)
    at Object.module.exports.raw [as parser] (/home/sobakasu/unisync/projects/autonano/node_modules/johnny-five/node_modules/serialport/parsers.js:8:13)
    at Object.SerialPort.opts.dataCallback (/home/sobakasu/unisync/projects/autonano/node_modules/johnny-five/node_modules/serialport/serialport.js:181:12)
    at SerialPortFactory.SerialPort._emitData (/home/sobakasu/unisync/projects/autonano/node_modules/johnny-five/node_modules/serialport/serialport.js:396:20)
```

## Code used to test

```js
var five = require("johnny-five");

five.Board().on("ready", function() {
  // This requires OneWire support using the ConfigurableFirmata
  var temperature = new five.Temperature({
    controller: "DS18B20",
    pin: 2
    // address: 0x3B8A761800000052
    // address: 0x3BF172180000002D
  });

  temperature.on("data", function() {
    console.log("0x" + this.address.toString(16));
    console.log(this.celsius + "°C", this.fahrenheit + "°F");
  });
});
```

BrianGenisio, 2015-09-09:
Sorry, I just saw this request to take a look.

Just because it is a OneWire device doesn't mean it will work out of the box with a different OneWire device.  

I took a quick look at the [data sheet](http://www.adafruit.com/datasheets/MAX31850-MAX31851.pdf) which suggests that it has a different "family code".  The MAX31850 has a family code of 0x3B where the DS18B20 has a family code of 0x28.  That alone will make it so the devices can't be found.  The rest of the interfaces are likely different.

I don't have any of these devices to test with, but I think the answer is that they are unsupported until someone writes the driver.  They are somewhat expensive, so I am reluctant to do it myself unless you want to ship some to me? I'd be happy to send them back.

talexand, 2015-09-11:
Thanks @BrianGenisio! Once we get in the analog replacements, I'll pester @bstein5 to send them over.

bstein5, 2015-10-11:
Hi Brian,
I'd like to mail you one of my Onewire MAX31850 thermocouple signal
amplifiers so you can figure out how to integrate it into the Johnny Five
framework. What mailing address would you prefer?
Thanks,
Ben

BrianGenisio, 2015-10-12:
@bstein5 Email me at BrianGenisio at gmail

BrianGenisio, 2015-10-20:
Ok, I have one of these amplifiers (Thanks Ben!) now.

I haven't hooked it up yet.  Just reading.  This worries me... apparently, they have a custom version of the  OneWire library!?!?  Not sure why.  Depending on why that is could make support for this device non-standard.  Not only do you need to include ConfigurableFirmata (instead of StandardFirmata), it has to include a custom OneWire lib?  

https://learn.adafruit.com/adafruit-1-wire-thermocouple-amplifier-max31850k/wiring-and-test

Still learning, but this could be a pretty big roadblock to supporting this device.  

Thoughts @bstein5 and @rwaldron ?

BrianGenisio, 2015-10-20:
Looking into this more, I'm skeptical that they modified the OneWire library.  I pulled it down and ran a diff on it.  I don't see any difference between what they call a custom OneWire library, and OneWire v2.2 .

Looks like it may be as easy as modifying the DS18B20 driver to support a different family code... Hoping to get it all hooked up tonight to start playing :)

I also think I can test this without a level converter because my [Arduino variant](http://www.sainsmart.com/sainsmart-uno-r3-atmega328-au-development-board-compatible-with-arduino-uno-r3.html) has a switch that can run in 3.3v I/O mode.  Do you think that is safe @bstein5?  Or should I rely on the level converter?  (If so, I'll need to wait a few days until I get into the office to grab one from the lab.)

bstein5, 2015-10-20:
I used a logic level converter in my setup, but I was going off what the
arduino website suggested, since I'm very new to the world of electrical
prototyping. If you can run the I/O on 3.3v then I'd say go for it, but
double check the data page on the arduino site, since as I recall it
specifically warns only to do that if you "know what you're doing". Either
way let me know how it shakes out.
Thanks!

BrianGenisio, 2015-10-24:
It took me a while to get the components I needed so I could set it up.  I decided to go with the level converter.  

A few observations:

1. I was able to get it working with the Arduino library example, but the probe doesn't seem to affect the reading.  There is an on-board temperature sensor which I can blow on and see the values change, but the probe itself doesn't seem to do anything.  Were you able to get this working with plain old Arduino?  I'm not sure if I'm doing something wrong... I dunno.

2. I was only able to get it to respond to the search command when it was in powered mode.  It doesn't seem to work in Parasitic power mode.  I'm not sure what to make of that, but whatevs.

3. Simply changing the family code was not enough to get this to work.  It gets past the detection stage, but then it just reads zeros from the board.  So it will take some more noodling to get it to work.  I'll dig in more in the coming days.

bstein5, 2015-10-24:
Not sure what to say, I had it working with just the arduino, both in
parasitic mode and powered. The only real issue I ran into was getting a
null signal due to a ground loop when I powered the other components, which
I was able to fix by electrically isolating the thermoprobes.
Perhaps its wired so the chip itself acts as the thermoprobe? Not sure how
to do that, but I remember reading about it on the website.  Can you send
me a picture of how you have it wired up?
-b

BrianGenisio, 2015-10-29:
[image]

This is how I have it wired.  The MAX31850 is connected to 5V (VIN) and ground.  The data pin has a pull-up resistor to 3.3 v and is connected to L1 on the level converter.  The H1 on the level converter is connected to digital 2.  The thermocouple is connected with the two red wires on the "red" terminal and the blue wire on the "yellow" terminal.

I think the wiring is correct, or else we wouldn't get readings... but I'm not sure (yet) why the thermocouple doesn't affect change the way the chip itself does.  My readings are quite accurate compared... so the data is flowing.

BrianGenisio, 2015-10-29:
Also, good news!  I have a very crude, but working example of reading the data from this sensor.  At least, it is reading just like the Arduino version of the code... from the chip, not the thermocouple.

There were only two modifications necessary.  The first is the family code.  This device has a family code of 0x3B.  The other change is that you need to poll the chip for data until it is ready.  A simple delay does not seem to be enough... at least a hardware delay doesn't cut it.  I modeled it after the Cpp code which polls the chip until it has data and boom!  Data started flowing.

So I'm going to play a bit more... can we just do a `setTimeout` based on the resolution?  Or do really need to read the scratch pad multiple times?  I don't know yet.  But I'm making progress, albeit a bit slow.     I haven't had a ton of time to look at it... If you want to take a stab, knowing what I've learned, let me know what you figure out.

bstein5, 2015-11-08:
Could you please post the code you've hashed out so far, especially the family code, so I can run it on my end and check? I've determined that analog signals are simply too finicky for the temperatures and setup I'm working with, so digital it is!

BrianGenisio, 2015-11-09:
@bstein5 Yeah.  Sorry I haven't gotten any further yet...  life has gotten in the way. This is not complete, though it works.  It needs work to be less copy-pasty.  It is mostly my hacking to get it working.  If you work on it further, let's collaborate.  Show me your code, etc.  Perhaps it makes sense for me to make you an admin on my fork?  Anyways, here is the start: https://github.com/BrianGenisio/johnny-five/commit/89eec79c8f4a3dd1bc861c31cb41df026d9c0d7a

bstein5, 2015-12-30:
> Anyways, here is the start: BrianGenisio@89eec79

Hi, just checking in. Have you been able to make any progress on the
drivers? Haven't seen any updates posted on Github.
I've been trying to determine the source of the problem, and it seems like
the program only crashes after the probe goes through a wide enough
temperature range. Since it works fine when running in the Arduino
environment, I figured the issue must be somewhere between translating
Arduino code to Javascript, and found in the Configurable Firmata code
(L306):
/* SEND STREAM WRITE BUFFER - TO DO: make sure that the stream buffer
doesn't go over
   * 60 bytes. use a timer to sending an event character every 4 ms to
   * trigger the buffer to dump. */

Could this be the source of the issue? Please look into it and let me know
your thoughts.
Thanks

BrianGenisio, 2015-12-30:
I haven't gotten a chance to look at it.  We're getting ready for a 200-person NodeBots workshop next Tuesday and that has consumed all of my NodeBots related time.  I haven't forgotten about it and hope to look at it soon.

As for the buffer issue... I really don't know.  Seems like a separate issue.

bstein5, 2015-12-30:
Understood. I don't want to be pushy, but I really wasn't expecting it to
take this long. I understand you're doing this in your spare time and I
appreciate what you've done thus far, but I have to ask if there is anyone
else you know or know of, who I could ask to finish this work while you're
occupied? I'm willing to pay, if it means finishing this in a timely
manner.
Please understand that I can't move forward with my project until this
issue is resolved, and I'm losing money every day I'm sitting still, since
I'm living off of savings. I don't believe there is much left to be done
with the sensor, but I do need it to be finished before I can move forward.
If there is an intern or someone at your office who I could hire to finish
coding the driver and testing the device, please let me know so I can
arrange something with them.
Let me know what you think.
-B

BrianGenisio, 2016-01-14:
Ok, I got a chance to take this out again tonight and get acclimated.  I was hoping to pull out some shared functionality between the two drivers to avoid duplication, but however I try, I find myself making it worse than it was before.  

So I'm going to give up on that for now.  Perhaps that can be a later refactor.

Here is what needs to be done to be complete:
1. The `isConversionAvailable` function needs to account for all devices, not just one.  I don't have two sensors. @bstein5 do you?  If I get this function modified, can you test it?
2. After that, I need to write some unit tests
3. Then a schematic and documentation

In all, I don't think it will take a ton more work.  I'm going to shoot for having the first task done by the end of the weekend (assuming you can test it with multiple devices).  Then I'll get the other tasks done shortly after that.  I think we are looking at being done in about a week unless something else comes up.  With the conference done, I'm setting this as my highest OSS/community task right now, and I have some others I'm itching to get to, so I'm on it ;)

soundanalogous, 2016-01-14:
There is also a chance that Firmata OneWire is failing. It is not a well tested feature and has not been supported for a few years now.

BrianGenisio, 2016-01-15:
@bstein5 I think I have a fully functional driver working on my branch: https://github.com/BrianGenisio/johnny-five/tree/max31850

Please try it out with multiple sensors and let me know how it goes.  It is working well with my one sensor, but it is now waiting for all the sensors to have finished the conversion before reading the values.

I'll get started on unit tests, now.

BrianGenisio, 2016-01-15:
@bstein5 Got unit tests passing.  I have a unit test with two devices, and it seems to work with stubbed data.  

I'll work on documentation in the next day or so.  Almost done.

reconbot, 2016-05-25:
@BrianGenisio were you able to finish this up?

BrianGenisio, 2016-05-25:
Yeah, that is wrapped up.  PR here: https://github.com/rwaldron/johnny-five/pull/1015

reconbot, 2016-05-25:
Awesome!

closed 2016-05-25 by reconbot