rwaldron/johnny-five #901: Error on install

henricavalcante, 2015-09-07:
````
[21:01:04] henricavalcante@machine:~/git/henricavalcante/node-aiopv2 (master)
$ npm install johnny-five
npm WARN package.json node-aiopv2@1.0.0 No description
npm WARN package.json node-aiopv2@1.0.0 No repository field.
npm WARN package.json node-aiopv2@1.0.0 No README data
\
> serialport@1.7.4 install /Users/henricavalcante/git/henricavalcante/node-aiopv2/node_modules/johnny-five/node_modules/serialport
> node-pre-gyp install --fallback-to-build

  CXX(target) Release/obj.target/serialport/src/serialport.o
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
../node_modules/nan/nan.h:261:25: error: redefinition of '_NanEnsureLocal'
NAN_INLINE v8::Local<T> _NanEnsureLocal(v8::Local<T> val) {
                        ^
../node_modules/nan/nan.h:256:25: note: previous definition is here
NAN_INLINE v8::Local<T> _NanEnsureLocal(v8::Handle<T> val) {
                        ^
../node_modules/nan/nan.h:661:13: error: no member named 'smalloc' in namespace 'node'
    , node::smalloc::FreeCallback callback
      ~~~~~~^
../node_modules/nan/nan.h:672:12: error: no matching function for call to 'New'
    return node::Buffer::New(v8::Isolate::GetCurrent(), data, size);
           ^~~~~~~~~~~~~~~~~
/Users/henricavalcante/.node-gyp/3.2.0/include/node/node_buffer.h:31:40: note: candidate function not viable: no known conversion from 'uint32_t' (aka 'unsigned int') to
      'enum encoding' for 3rd argument
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
/Users/henricavalcante/.node-gyp/3.2.0/include/node/node_buffer.h:43:40: note: candidate function not viable: 2nd argument ('const char *') would lose const qualifier
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
/Users/henricavalcante/.node-gyp/3.2.0/include/node/node_buffer.h:28:40: note: candidate function not viable: requires 2 arguments, but 3 were provided
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate, size_t length);
                                       ^
/Users/henricavalcante/.node-gyp/3.2.0/include/node/node_buffer.h:36:40: note: candidate function not viable: requires 5 arguments, but 3 were provided
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
../node_modules/nan/nan.h:676:12: error: no viable conversion from 'v8::MaybeLocal<v8::Object>' to 'v8::Local<v8::Object>'
    return node::Buffer::New(v8::Isolate::GetCurrent(), size);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/henricavalcante/.node-gyp/3.2.0/include/node/v8.h:210:7: note: candidate constructor (the implicit copy constructor) not viable: no known conversion from
      'v8::MaybeLocal<v8::Object>' to 'const v8::Local<v8::Object> &' for 1st argument
class Local {
      ^
/Users/henricavalcante/.node-gyp/3.2.0/include/node/v8.h:210:7: note: candidate constructor (the implicit move constructor) not viable: no known conversion from
      'v8::MaybeLocal<v8::Object>' to 'v8::Local<v8::Object> &&' for 1st argument
class Local {
      ^
/Users/henricavalcante/.node-gyp/3.2.0/include/node/v8.h:214:13: note: candidate template ignored: could not match 'Local' against 'MaybeLocal'
  V8_INLINE Local(Local<S> that)
            ^
/Users/henricavalcante/.node-gyp/3.2.0/include/node/v8.h:327:13: note: candidate template ignored: could not match 'S *' against 'v8::MaybeLocal<v8::Object>'
  V8_INLINE Local(S* that)
            ^
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
../node_modules/nan/nan.h:683:26: error: no member named 'Use' in namespace 'node::Buffer'
    return node::Buffer::Use(v8::Isolate::GetCurrent(), data, size);
           ~~~~~~~~~~~~~~^
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
In file included from ../node_modules/nan/nan.h:24:
In file included from /Users/henricavalcante/.node-gyp/3.2.0/include/node/node.h:42:
/Users/henricavalcante/.node-gyp/3.2.0/include/node/v8.h:221:5: error: assigning to 'v8::Primitive *volatile' from incompatible type 'v8::Value *'
    TYPE_CHECK(T, S);
    ^~~~~~~~~~~~~~~~
/Users/henricavalcante/.node-gyp/3.2.0/include/node/v8.h:180:37: note: expanded from macro 'TYPE_CHECK'
[... 76 lines cut ...]
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
node-pre-gyp ERR! System Darwin 14.5.0
node-pre-gyp ERR! command "/usr/local/bin/iojs" "/Users/henricavalcante/git/henricavalcante/node-aiopv2/node_modules/johnny-five/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /Users/henricavalcante/git/henricavalcante/node-aiopv2/node_modules/johnny-five/node_modules/serialport
node-pre-gyp ERR! node -v v3.2.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.7
node-pre-gyp ERR! not ok
Failed to execute '/usr/local/bin/iojs /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/Users/henricavalcante/git/henricavalcante/node-aiopv2/node_modules/johnny-five/node_modules/serialport/build/serialport/v1.7.4/Release/node-v45-darwin-x64/serialport.node --module_name=serialport --module_path=/Users/henricavalcante/git/henricavalcante/node-aiopv2/node_modules/johnny-five/node_modules/serialport/build/serialport/v1.7.4/Release/node-v45-darwin-x64' (1)
npm WARN optional dep failed, continuing serialport@1.7.4
johnny-five@0.8.90 node_modules/johnny-five
├── ease-component@1.0.0
├── descriptor@0.1.0
├── color-convert@0.5.3
├── es6-shim@0.33.3
├── nanotimer@0.3.10
├── temporal@0.4.1
├── lodash@3.10.1
├── chalk@1.1.1 (escape-string-regexp@1.0.3, supports-color@2.0.0, ansi-styles@2.1.0, strip-ansi@3.0.0, has-ansi@2.0.0)
├── array-includes@3.0.1 (define-properties@1.1.1, es-abstract@1.3.1)
└── firmata@0.5.5 (browser-serialport@2.0.3, object-assign@1.0.0, es6-map@0.1.1)

[21:03:17] henricavalcante@machine:~/git/henricavalcante/node-aiopv2 (master)
$ node -v
v3.2.0
````

fivdi, 2015-09-07:
The addon serialport hasn't been updated to support io.is/Node.js v3.x yet. There are PRs for adding the required support.

rwaldron, 2015-09-07:
Thanks for filing the issue, I will close this in favor of the [thread on node-serialport](https://github.com/voodootikigod/node-serialport/issues/573#issuecomment-138259676)

closed 2015-09-07 by rwaldron

henricavalcante, 2015-09-07:
Thanks guys :)

wangkewei1993, 2020-02-18:
说一下我在安装johnny-five 和seriosport时各种问题的解决办法,网上的各种方法都尝试了还是无法安装,最后我安装好环境之后重启了一下电脑,并将node版本切换到8.11.3,安装成功!