rwaldron/johnny-five #902: Question about arduino network shield and touchscreen

Chris2011, 2015-09-10:
I didn't test it yet but I will, only what I wanted to know, is there support for these 2 shields?

Touchscreen with SD for Arduino - http://www.amazon.de/Foxnovo-Touchscreen-SD-Kartenschacht-Arduino-Board/dp/B00P5QJHGC

Ethernet Shield w5100 - http://www.amazon.de/Ethernet-Schild-shield-Arduino-w5100/dp/B009N826DY

Thx in advance. :)

rwaldron, 2015-09-10:
@Chris2011 hi! I'm on vacation right now and won't be back until this time next week. Someone else might jump in before then, but I didn't want to leave you hanging until then

Chris2011, 2015-09-10:
Thx for your answer. Happy vacation :). Only one thing, that I understand it right, I can't put the JS script onto the arduino and get it working without a host computer right? It is only working via connection to a host computer, I use nodejs to write stuff and communicate with arduino over the firmata protocoll? So it is not working with a standalone arduino board + modules, right?

fivdi, 2015-09-10:
>I can't put the JS script onto the arduino and get it working without a host computer right?

Correct

>It is only working via connection to a host computer, I use nodejs to write stuff and communicate with arduino over the firmata protocoll?

Correct

>So it is not working with a standalone arduino board + modules, right?

Correct

rwaldron, 2015-09-10:
Inline...

> Thx for your answer. Happy vacation :). Only one thing, that I understand
> it right, I can't put the JS script onto the arduino and get it working
> without a host computer right? It is only working [...]

Correct, but you _can_ run the JS code on platforms like:

- Tessel 2
- Intel Edison (which has an Arduino compatible breakout)
- Intel Galileo Gen 1 & 2
- pcDuino 3
- Raspberry Pi
- BeagleBone Black
- Linino One

The whole list, plus support info, is here:
http://johnny-five.io/platform-support/#relationship:embedded

Rick

Chris2011, 2015-09-10:
Ok Thx, I only have an Arduino Uno R3, I like the language but I thought that could be easier to use JS. I like the principle behind that to use JS for the IoT AFAP. SO I thought that this here is working directly on a standalone Arduino without a host computer. Anyway, I will try the Tessel 2 soon. Thx for your help and enjoy your vacation.

Cheers

Chris

dtex, 2016-01-23:
@Chris2011 Okay to close this?

Chris2011, 2016-01-23:
Yes we can :)

closed 2016-01-23 by dtex