rwaldron/johnny-five #905: Support for 7-segment LED HT16K33 Backpack

donmccurdy, 2015-09-14:
Hi there! I have an [Adafruit 4-digit 7-segment display w/ I2C backpack](http://www.adafruit.com/products/879), and have been struggling to get it working. Originally I thought I could use `five.Led.Digits`, but the `controller: "HT16K33"` option seems to force the type to `five.Led.Matrix`. I dug into ledcontrol.js a bit and it looks like HT16K33 is assumed to be a matrix-only controller.

I finally did get things working:

```javascript
var five = require('johnny-five');
var Raspi = require('raspi-io');
var board = new five.Board({io: new Raspi()});

var digits = [0x3F, 0x06, 0x5B, 0x4F, 0x66, 0x6D, 0x7D, 0x07, 0x7F, 0x6F];

board.on("ready", function() {
    // Create HT16K33 controller
    var matrix = new five.Led.Matrix({controller: 'HT16K33'});

    // Display 12:34
    matrix.displaybuffers[0][0] = digits[1];
    matrix.displaybuffers[0][1] = digits[2];
    matrix.displaybuffers[0][2] = 0xFFFF;
    matrix.displaybuffers[0][3] = digits[3];
    matrix.displaybuffers[0][4] = digits[4];
    matrix.writeDisplay(0);
});
```

The digits are from an [Adafruit python tutorial](https://github.com/adafruit/Adafruit-Raspberry-Pi-Python-Code/blob/master/Adafruit_LEDBackpack/Adafruit_7Segment.py).

I'm assuming there must be a better way to do this. Would you take a PR implementing `digit()` and `print()` for the HT16K33 controller, or would that even be the right approach? I skimmed through #665, which sounded like it would address my question as well.

Thanks!

rwaldron, 2015-09-16:
> looks like HT16K33 is assumed to be a matrix-only controller.

That's correct, it's just a matter of someone doing the work to add support. I have a couple of these, I just haven't had time to work on it. 

> I'm assuming there must be a better way to do this.

Yep—once I or anyone else that wants to work on it has a chance to add support, then it will be as easy as the API shown in use here: http://johnny-five.io/examples/led-digits-clock-galileo/

donmccurdy, 2015-09-17:
Thanks Rick – I guess what I should have asked then, if wrote up a PR implementing the HT16K33 controller's [digits()](https://github.com/rwaldron/johnny-five/blob/master/lib/led/ledcontrol.js#L416-L419) and [print()](https://github.com/rwaldron/johnny-five/blob/master/lib/led/ledcontrol.js#L504-L507) methods, would that be a reasonable way to add support? Or is there some larger refactor of ledcontrol.js that you'd want to see first, based on comments in the other issue?

rwaldron, 2015-09-17:
>  if I wrote up a PR implementing the HT16K33 controller's digits() and print() methods, would that be a reasonable way to add support?

Yes. 

> is there some larger refactor of ledcontrol.js that you'd want to see first, based on comments in the other issue?

There is, but I wouldn't consider it a blocker here. It just means that somewhere down the line I will find you and ask you to assist in reviewing/contributing to the refactoring.

PR #907 (2015-09-30): Support Led.Digits with HT16K33 controller
Fixes #905.

Moves print() and digits() to parent controller, calling into delegated sendDigit() functions for controller-specific logic.

There were a few differences with the HT16K33 controller:

1. Column indexing is left-to-right, so I've added a `digitsOrder` property to the controller. I'm not sure which direction is actually more common.
2. The order of individual segments is reversed, so the constants in [DIGIT_CHARS](https://github.com/rwaldron/johnny-five/blob/master/lib/led/led-chars.js#L296) don't work. Rather than adding another set of co [... cut]
closed 2015-09-30 by rwaldron