rwaldron/johnny-five #909: arduino ad9850

kukulako, 2015-09-24:
can i use this script whit arduino ad9850 signal generator.....?/

rwaldron, 2015-11-05:
Sorry for the delay in responding. Currently there is no explicit support for such devices, but johnny-five is always interested in exploring possibilities.

kukulako, 2015-11-06:
ok thanks..

kukulako, 2015-11-07:
@rwaldron  

this is a arduino sketch in C
---------------------------------------------------------------------------------------------------------------------------
/*
 * A simple single freq AD9850 Arduino test script
 * Original AD9851 DDS sketch by Andrew Smallbone at www.rocketnumbernine.com
 * Modified for testing the inexpensive AD9850 ebay DDS modules
 * Pictures and pinouts at nr8o.dhlpilotcentral.com
 * 9850 datasheet at http://www.analog.com/static/imported-files/data_sheets/AD9850.pdf
 * Use freely
 */
 
 #define W_CLK 8       // Pin 8 - connect to AD9850 module word load clock pin (CLK)
 #define FQ_UD 9       // Pin 9 - connect to freq update pin (FQ)
 #define DATA 10       // Pin 10 - connect to serial data load pin (DATA)
 #define RESET 11      // Pin 11 - connect to reset pin (RST).
 
 #define pulseHigh(pin) {digitalWrite(pin, HIGH); digitalWrite(pin, LOW); }
 
 // transfers a byte, a bit at a time, LSB first to the 9850 via serial DATA line
void tfr_byte(byte data)
{
  for (int i=0; i<8; i++, data>>=1) {
    digitalWrite(DATA, data & 0x01);
    pulseHigh(W_CLK);   //after each bit sent, CLK is pulsed high
  }
}
 
 // frequency calc from datasheet page 8 = <sys clock> * <frequency tuning word>/2^32
void sendFrequency(double frequency) {
  int32_t freq = frequency * 4294967295/125000000;  // note 125 MHz clock on 9850
  for (int b=0; b<4; b++, freq>>=8) {
    tfr_byte(freq & 0xFF);
  }
  tfr_byte(0x000);   // Final control byte, all 0 for 9850 chip
  pulseHigh(FQ_UD);  // Done!  Should see output
}
 
void setup() {
 // configure arduino data pins for output
  pinMode(FQ_UD, OUTPUT);
  pinMode(W_CLK, OUTPUT);
  pinMode(DATA, OUTPUT);
  pinMode(RESET, OUTPUT);
 
  pulseHigh(RESET);
  pulseHigh(W_CLK);
  pulseHigh(FQ_UD);  // this pulse enables serial mode - Datasheet page 12 figure 10
}
 
void loop() {
  sendFrequency(10.e6);  // freq
  while(1);
}

 someone hacked that Arduino sketch into a piece of Python code for the Pi..
-----------------------------------------------------------------------------------------------------------------------------------------

#!/usr/local/bin/python
# Run an eBay AD9850 on the RPi GPIO
#
# translated from nr8o's Arduino sketch
# at http://nr8o.dhlpilotcentral.com/?p=83
# 
# m0xpd
# shack.nasties 'at Gee Male dot com'
import RPi.GPIO as GPIO
# setup GPIO options...
GPIO.setmode(GPIO.BOARD)
GPIO.setwarnings(False)

W_CLK = 12              			# Define GPIO pins
FQ_UD = 16
DATA = 18
RESET = 22  

def pulseHigh(pin):               		# Function to send a pulse
	GPIO.output(pin, True)          	# do it a few times to increase pulse width
	GPIO.output(pin, True)          	# (easier than writing a delay loop!)
	GPIO.output(pin, True)
	GPIO.output(pin, False)         	# end of the pulse
	return

def tfr_byte(data):               		# Function to send a byte by serial "bit-banging"
	for i in range (0,8):
		GPIO.output(DATA, data & 0x01)	# Mask out LSB and put on GPIO pin "DATA"
		pulseHigh(W_CLK)              	# pulse the clock line
		data=data>>1                  	# Rotate right to get next bit
	return

def sendFrequency(frequency):     		# Function to send frequency (assumes 125MHz xtal)
	freq=long(frequency*4294967296/125000000)
	for b in range (0,4):
		tfr_byte(freq & 0xFF)
		freq=freq>>8
	tfr_byte(0x00)
	pulseHigh(FQ_UD)
	return

GPIO.setup(W_CLK, GPIO.OUT)       		# setup IO bits...
GPIO.setup(FQ_UD, GPIO.OUT)       		#
GPIO.setup(DATA, GPIO.OUT)        		#
GPIO.setup(RESET, GPIO.OUT)       		#

GPIO.output(W_CLK, False)         		# initialize everything to zero...
GPIO.output(FQ_UD, False)
GPIO.output(DATA, False)
GPIO.output(RESET, False)

pulseHigh(RESET)                  		# start-up sequence...
pulseHigh(W_CLK)
pulseHigh(FQ_UD)

frequency = 7000000               		# choose frequency and
sendFrequency(frequency)          		# start the oscillator

web pages credits

http://m0xpd.blogspot.com/2013_01_01_archive.html
http://nr8o.dhlpilotcentral.com/?p=83

closed 2016-05-25 by reconbot