rwaldron/johnny-five #910: Question about Heroku Deployment

jacklee0223, 2015-09-28:
Hi,

I am a recent grad at General Assembly's WDI, meaning I am still a complete noob at any web development. So please excuse me for any incompetency.

For my final project, I have chose to make an app that allows users to control drones via browser. On top of Node.js and Johnny-Five, I've used Express, Angular, and Socket.io to allow the control of the drones from the DOM elements on the browser.

However, when I deploy it on heroku, the app crashes and gives me the following message:

```
2015-09-28T07:25:03.911098+00:00 heroku[web.1]: State changed from crashed to starting
2015-09-28T07:25:10.766738+00:00 heroku[web.1]: Starting process with command `node server.js`
2015-09-28T07:25:13.654291+00:00 app[web.1]: Waiting for connection
2015-09-28T07:25:13.582913+00:00 app[web.1]: port 21480
2015-09-28T07:25:13.658160+00:00 app[web.1]:       ports = result.filter(function(val) {
2015-09-28T07:25:13.658161+00:00 app[web.1]:                     ^
2015-09-28T07:25:13.658164+00:00 app[web.1]:     at Board.<anonymous> (/app/node_modules/johnny-five/lib/board.js:83:21)
2015-09-28T07:25:13.658163+00:00 app[web.1]: TypeError: Cannot read property 'filter' of undefined
2015-09-28T07:25:13.658165+00:00 app[web.1]:     at /app/node_modules/serialport/serialport.js:565:11
2015-09-28T07:25:13.658166+00:00 app[web.1]:     at FSReqWrap.oncomplete (fs.js:95:15)
2015-09-28T07:25:13.658158+00:00 app[web.1]: /app/node_modules/johnny-five/lib/board.js:83
2015-09-28T07:25:14.645698+00:00 heroku[web.1]: Process exited with status 1
```

I am thinking several possibilities for this happening:
1. Since the Node server on my heroku app is on cloud, whereas Node server on my local server is on my computer, it cannot locate my device
2. Since Serialport requires xCode CLI, I need to add plugin on Heroku (already done, still not working but maybe I am missing a step?)
3. Johnny-Five is not designed to be deployed on web

I know my question is not very clear, but I would appreciate any input.

Github link: https://github.com/jacklee0223/hyperion
Heroku link: https://young-crag-5724.herokuapp.com/

*For the Heroku, I've commented out lines related to Johnny-Five on server.js in order for the app not to crash. Of course, the drone control does not work on Heroku app.

Thanks.

BrianGenisio, 2015-09-28:
Hey there!  Sounds like an exiting project you are working on!

From what I understand of your project, you are using Bluetooth or Serial (USB) to connect to an Arduino.  The moment the code goes to Heroku, the code is running somewhere in the cloud, completely disconnected from your board.  In its current state, you'd need to connect the Arduino into a computer some place in a data center in some unknown city in some unknown country ;)  But that isn't happening, really, because Heroku servers are running on virtual machines without any physical Bluetooth or USB connections.

Unfortunately, this isn't going to work as-is.  You need another layer of abstraction.  Your web app will need to communicate with a local agent some place which is has a connection to your board.  You might be able to accomplish this via proxy (using [remote-io](https://github.com/monteslu/remote-io).  It might require some playing and a public IP address for the agent and it _might_ work.

Honestly, though, this seems like a job for [Zetta](http://www.zettajs.org/), which is a good IoT layer that sits on top of hardware.  You can still use Johnny-Five to abstract the hardware, but Zett's job is to get the data from an agent like yours to and from a cloud location (like Heroku).  It has really great dashboards and a great REST API.  

Good luck!

jacklee0223, 2015-09-29:
Thank you for your thorough explanation!

It all makes perfect sense now and I don't know whether I should be relieved to know what's happening or bummed there is no way to make it work as-is (both but leaning more towards the latter).

Zetta sounds like an interesting platform and I will definitely spend my time learning more about it.

Thanks!

closed 2015-09-29 by jacklee0223

rafagarciaarribas, 2016-10-15:
Hi @jacklee0223,

I arrived because I´m doing a project very similar to yours, using Node, Angular, Johnny-five to communicate with Arduino and show info from sensor on the client.

Did you resolve your problems with Zetta? Or could you give me any recommendation?

Thank you!

I tried as you to deploy on heroku with a this error:

<pre><code>2016-10-14T18:33:30.537775+00:00 app[web.1]: Error: /app/node_modules/johnny-fi
e/node_modules/serialport/build/Release/serialport.node: invalid ELF header
2016-10-14T18:33:30.537770+00:00 app[web.1]:
2016-10-14T18:33:30.537777+00:00 app[web.1]:     at Object.Module._extensions..
ode (module.js:440:18)
2016-10-14T18:33:30.537778+00:00 app[web.1]:     at Module.load (module.js:357:2)
2016-10-14T18:33:30.537778+00:00 app[web.1]:     at Function.Module._load (module.js:314:12)
2016-10-14T18:33:30.537776+00:00 app[web.1]:     at Error (native)
2016-10-14T18:33:30.537779+00:00 app[web.1]:     at Module.require (module.js:3
7:17)
2016-10-14T18:33:30.537779+00:00 app[web.1]:     at require (internal/module.js
20:19)
2016-10-14T18:33:30.537780+00:00 app[web.1]:     at bindings (/app/node_modules
johnny-five/node_modules/serialport/node_modules/bindings/bindings.js:76:44)
2016-10-14T18:33:30.537781+00:00 app[web.1]:     at Object.<anonymous> (/app/no
e_modules/johnny-five/node_modules/serialport/lib/bindings.js:3:35)
2016-10-14T18:33:30.537781+00:00 app[web.1]:     at Module._compile (module.js:
13:34)
2016-10-14T18:33:30.537782+00:00 app[web.1]:     at Object.Module._extensions..
s (module.js:422:10)
2016-10-14T18:33:30.641489+00:00 heroku[web.1]: Process exited with status 1
2016-10-15T00:09:32.111685+00:00 heroku[web.1]: State changed from crashed to s
arting
2016-10-15T00:09:33.679588+00:00 heroku[web.1]: Starting process with command `
ode server.js`
2016-10-15T00:09:35.727529+00:00 app[web.1]: Server available at http://localho
t:3000
2016-10-15T00:09:35.748851+00:00 app[web.1]: /app/node_modules/johnny-five/node
modules/serialport/node_modules/bindings/bindings.js:83
2016-10-15T00:09:35.748858+00:00 app[web.1]:         throw e

</code></pre>