rwaldron/johnny-five #911: Nodejs crashes on HTTP request

cpustar, 2015-09-28:
```javascript
var five = require("johnny-five");
var board = new five.Board({
  repl: false
});

board.on("ready", function() {
  // Create an Led on pin 13
  var led = new five.Led(13);
  // Blink every half second
  led.blink(500); 
});
```
The code works if I don't make any HTTP requests.

As soon as I make a GET request it crashes, for example: http://localhost:3000 

Node.js version: v4.1.0
Serialport version: github.com/ghostoy/node-serialport.git#nan1to2
Express version: 4.13.1
johnny-five: 0.8.92

Board: Arduino Uno

UlisesGascon, 2015-09-29:
Hi! As your script needs Arduino, you can always add the HTTP part inside the Board callback as follow:

```javascript
var express = require('express');
var app = express();
var five = require("johnny-five");
var board = new five.Board({
  repl: false
});

// We wait until the board is ready.
board.on("ready", function() {
	// Express
	app.get('/', function (req, res) {
	  res.send('Hello World!');
	});
	var server = app.listen(3000, function () {
	  var host = server.address().address;
	  var port = server.address().port;

	  console.log('Example app listening at http://%s:%s', host, port);
	});

	// Arduino
	var led = new five.Led(13);
	led.blink(500); 
});
```

cpustar, 2015-09-29:
@UlisesGascon 

Thanks for responding. It was an issue with Node  v4.1.0.

I reverted back to node v0.12.6 and everything works fine.

closed 2015-09-29 by cpustar