rwaldron/johnny-five #913: Unable to install johnny five anymore

protozoo, 2015-10-02:
Hi there,

this past week i've been testing johnny five without problems, untill yesterday, when I tried to install-it in an [electron](http://electron.atom.io) project. As it didn't work out of the box, I've been messing around following partial solutions taken from many places (like [here](https://gist.github.com/jedthehumanoid/a7f8278e0a37d259adca), [here](https://github.com/jgautier/firmata/issues/96), [here](https://www.npmjs.com/package/serialport-electron) and some more). None of them finally worked and I finally got to a point where trying to reinstall johnny five (locally or globally) threw a big error.

So I decided to uninstall and reinstall Node, which I just did (I'm on Node 4.1.1 now). However, I still can't get johnny-five installed. Here's my CLI output:

```
➜  node_mods_test  npm install serialport --save
npm WARN package.json node_mods_test@1.0.0 No description
npm WARN package.json node_mods_test@1.0.0 No repository field.
npm WARN package.json node_mods_test@1.0.0 No README data
\
> serialport@1.7.4 install /Users/daniel/temp/node_mods_test/node_modules/serialport
> node-pre-gyp install --fallback-to-build

  CXX(target) Release/obj.target/serialport/src/serialport.o
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
../node_modules/nan/nan.h:261:25: error: redefinition of '_NanEnsureLocal'
NAN_INLINE v8::Local<T> _NanEnsureLocal(v8::Local<T> val) {
                        ^
../node_modules/nan/nan.h:256:25: note: previous definition is here
NAN_INLINE v8::Local<T> _NanEnsureLocal(v8::Handle<T> val) {
                        ^
../node_modules/nan/nan.h:661:13: error: no member named 'smalloc' in namespace
      'node'
    , node::smalloc::FreeCallback callback
      ~~~~~~^
../node_modules/nan/nan.h:672:12: error: no matching function for call to 'New'
    return node::Buffer::New(v8::Isolate::GetCurrent(), data, size);
           ^~~~~~~~~~~~~~~~~
/Users/daniel/.node-gyp/4.1.1/include/node/node_buffer.h:31:40: note: candidate
      function not viable: no known conversion from 'uint32_t'
      (aka 'unsigned int') to 'enum encoding' for 3rd argument
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
/Users/daniel/.node-gyp/4.1.1/include/node/node_buffer.h:43:40: note: candidate
      function not viable: 2nd argument ('const char *') would lose const
      qualifier
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
/Users/daniel/.node-gyp/4.1.1/include/node/node_buffer.h:28:40: note: candidate
      function not viable: requires 2 arguments, but 3 were provided
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate, size_t length);
                                       ^
/Users/daniel/.node-gyp/4.1.1/include/node/node_buffer.h:36:40: note: candidate
      function not viable: requires 5 arguments, but 3 were provided
NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                       ^
In file included from ../src/serialport.cpp:3:
In file included from ../src/serialport.h:5:
../node_modules/nan/nan.h:676:12: error: no viable conversion from
      'v8::MaybeLocal<v8::Object>' to 'v8::Local<v8::Object>'
    return node::Buffer::New(v8::Isolate::GetCurrent(), size);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/daniel/.node-gyp/4.1.1/include/node/v8.h:210:7: note: candidate
      constructor (the implicit copy constructor) not viable: no known
      conversion from 'v8::MaybeLocal<v8::Object>' to 'const
      v8::Local<v8::Object> &' for 1st argument
class Local {
      ^
/Users/daniel/.node-gyp/4.1.1/include/node/v8.h:210:7: note: candidate
      constructor (the implicit move constructor) not viable: no known
      conversion from 'v8::MaybeLocal<v8::Object>' to 'v8::Local<v8::Object> &&'
      for 1st argument
class Local {
      ^
/Users/daniel/.node-gyp/4.1.1/include/node/v8.h:214:13: note: candidate template
      ignored: could not match 'Local' against 'MaybeLocal'
  V8_INLINE Local(Local<S> that)
[... 115 lines cut ...]
npm ERR! Darwin 14.0.0
npm ERR! argv "/usr/local/bin/node" "/Users/daniel/.node/bin/npm" "install" "serialport" "--save"
npm ERR! node v4.1.1
npm ERR! npm  v2.8.3
npm ERR! code ELIFECYCLE

npm ERR! serialport@1.7.4 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@1.7.4 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /Users/daniel/temp/node_mods_test/npm-debug.log
```

Any idea what's wrong?

Many thanks!

dtex, 2015-10-02:
Serialport does currently support 4.x The team is working on it at here:

https://github.com/voodootikigod/node-serialport/issues/578

I believe a major version Serialport update will be released any day now.

If you don't want to wait, there are two forks in that issue that you could install right now or you could roll back to node 0.12.7

closed 2015-10-02 by dtex

protozoo, 2015-10-02:
Thank you @dtex, I rolled back to node 0.12.7 and it's working again