rwaldron/johnny-five #915: support for Grove Q Touch

oallouch, 2015-10-03:
I'm quite a rookie, so maybe it's already there, but do you support the Q Touch's AT42QT107 ?
It talks I2C, so maybe it's simple to translate it from https://github.com/Seeed-Studio/Seeed_QTouch 

Thx a lot in advance,

Olivier

rwaldron, 2015-10-03:
Not currently supported, but looks trivial. I'll get one ordered. The plan:

- add an Expander controller for qtouch's chip
- add a Keypad controller that's implemented with the Expander instance

oallouch, 2015-10-03:
Keypad I saw, but Expander I'll learn.
(so much to learn, but it's very exciting)

oallouch, 2015-10-03:
Ok, I managed to get some data using i2cReadOnce.

	var address = 0x1B;
	this.i2cConfig();
	var self = this;
	setInterval(function() {
		self.i2cReadOnce(address, 0X03, 1, function(bytes) {
			console.log("Bytes read: ", bytes);
		});
	}, 100);
But it's way less elegant than the Expander/KeyPad solution.
If you got one, you'll still be my hero.

rwaldron, 2015-10-04:
Yes, I ordered the sensor yesterday :)

In the meantime, you're better off using `i2cRead` 

```ks
var address = 0x1B;
this.i2cConfig();
this.i2cRead(address, 0X03, 1, function(state) {
  for (var i = 0; i < 7; i++) {
    if (state & (1 << i)) {
      console.log("Pressed: %d", i);
    }
  }
});
```

I'll update you when my sensor arrives

oallouch, 2015-10-05:
This sensor is fun, even for kids games.

Thx a lot !

rwaldron, 2015-10-05:
@oallouch looking forward to playing with it :) Did my modified code work ok for getting all the key touches?

oallouch, 2015-10-06:
When I first tried, i2cRead didn't work. I should try again.

rwaldron, 2015-10-25:
My Grove Touch sensors finally arrived!!

oallouch, 2015-10-27:
So much fun

topkoa, 2015-10-31:
Hey has anyone messed with this sensor much yet?  

I got one to mess with, and am using the rwaldron i2c code above-- it seems to communicate OK-- however it seems WAY sensitive, such that when I leave it alone it's fine, but if I connect a wire to one of the sensor leads it continually reports a touch sense.  (as in insulated wire, without me touching it afterwards)

It's not much fun, if the act of hooking this thing to something triggers the touch before you actually touch the thing you hooked up.. *puzzled*

Thanks!

topkoa, 2015-10-31:
Duh.  I messed with the "delay" in the i2cConfig() and was able to dial it in.

oallouch, 2015-11-02:
You can also try (I haven't) to calibrate it using the registry 0x38 .
See https://github.com/Seeed-Studio/Seeed_QTouch/blob/master/Seeed_QTouch.h

topkoa, 2015-11-02:
Thanks so much!   This makes so much more sense... (I'm quite new to the concepts of i2c so this didn't occur to me..)

After reading that document; it's nice to know there is a "calibration" and "reset" for this chip-- in my testing, even after ratcheting up the delay to get initially better results, it would tend to "go off the rails" if there were too many touch events...  My (admittedly novice) mindset was not sure how to get back to square one without completely stopping and restarting my program...

rwaldron, 2015-11-03:
The code above is a raw reading and not actually what I expect anyone to use. I landed a new Keypad controller on master

rwaldron, 2015-11-03:
I'll push out a release tomorrow morning

rwaldron, 2015-11-03:
Also, about i2c delay: that's for creating _microsecond_ delay between setting the register and reading it. That's **process blocking** on the board, so don't do that ;)

rwaldron, 2015-11-03:
I just pushed the release to npm.

closed 2015-11-03 by rwaldron

rwaldron, 2015-11-03:
https://github.com/rwaldron/johnny-five/blob/master/docs/keypad-QTOUCH.md

oallouch, 2015-11-03:
You're the best.

Thx again

topkoa, 2015-11-04:
Seconded!

Oh, and thanks for setting me straight on my incorrect usage of the "delay"... I'm still learning the ropes on this stuff, and it's refreshing to get such helpful feedback!

rwaldron, 2015-11-04:
> Oh, and thanks for setting me straight on my incorrect usage of the "delay"... I'm still learning the ropes on this stuff, and it's refreshing to get such helpful feedback!

No problem! 

Anytime you have questions, or need any guidance, _please_ join us in the [Johnny-Five gitter](https://gitter.im/rwaldron/johnny-five). Everyone there is super welcoming and always willing to help or to point you to the right person for help!