rwaldron/johnny-five #920: Install J5 and Raspi-io on Raspbian Jessie

rudemex, 2015-10-09:
errors.

fivdi, 2015-10-09:
Please post the complete output of the install process.

rudemex, 2015-10-09:
pi@woonkypi ~ $ sudo npm install raspi-io
 
> raspi-wiringpi@1.0.6 install /home/pi/node_modules/raspi-io/node_modules/raspi/node_modules/raspi-wiringpi
> sh install-wiringpi.sh

Wiring Pi is already installed, skipping

 
> raspi-pwm@1.3.3 install /home/pi/node_modules/raspi-io/node_modules/raspi-pwm
> node-gyp rebuild

gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/4.0.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/node_modules/raspi-io/node_modules/raspi-pwm/.node-gyp"
make: Entering directory '/home/pi/node_modules/raspi-io/node_modules/raspi-pwm/build'
  CXX(target) Release/obj.target/addon/src/addon.o
In file included from ../node_modules/nan/nan_new.h:190:0,
                 from ../node_modules/nan/nan.h:74,
                 from ../src/addon.cc:26:
../node_modules/nan/nan_implementation_12_inl.h: In static member function 'static NanIntern::FactoryBase<v8::Signature>::return_t NanIntern::Factory<v8::Signature>::New(NanIntern::Factory<v8::Signature>::FTH, int, NanIntern::Factory<v8::Signature>::FTH*)':
../node_modules/nan/nan_implementation_12_inl.h:181:76: error: no matching function for call to 'v8::Signature::New(v8::Isolate*, NanIntern::Factory<v8::Signature>::FTH&, int&, NanIntern::Factory<v8::Signature>::FTH*&)'
   return v8::Signature::New(v8::Isolate::GetCurrent(), receiver, argc, argv);
                                                                            ^
../node_modules/nan/nan_implementation_12_inl.h:181:76: note: candidate is:
In file included from /home/pi/node_modules/raspi-io/node_modules/raspi-pwm/.node-gyp/4.0.0/include/node/node.h:42:0,
                 from ../src/addon.cc:25:
/home/pi/node_modules/raspi-io/node_modules/raspi-pwm/.node-gyp/4.0.0/include/node/v8.h:4675:27: note: static v8::Local<v8::Signature> v8::Signature::New(v8::Isolate*, v8::Local<v8::FunctionTemplate>)
   static Local<Signature> New(
                           ^
/home/pi/node_modules/raspi-io/node_modules/raspi-pwm/.node-gyp/4.0.0/include/node/v8.h:4675:27: note:   candidate expects 2 arguments, 4 provided
In file included from ../src/addon.cc:26:0:
../node_modules/nan/nan.h: At global scope:
../node_modules/nan/nan.h:165:25: error: redefinition of 'template<class T> v8::Local<T> _NanEnsureLocal(v8::Local<T>)'
 NAN_INLINE v8::Local<T> _NanEnsureLocal(v8::Local<T> val) {
                         ^
../node_modules/nan/nan.h:160:25: note: 'template<class T> v8::Local<T> _NanEnsureLocal(v8::Handle<T>)' previously declared here
 NAN_INLINE v8::Local<T> _NanEnsureLocal(v8::Handle<T> val) {
                         ^
../node_modules/nan/nan.h:557:13: error: 'node::smalloc' has not been declared
     , node::smalloc::FreeCallback callback
             ^
../node_modules/nan/nan.h:557:35: error: expected ',' or '...' before 'callback'
     , node::smalloc::FreeCallback callback
                                   ^
../node_modules/nan/nan.h: In function 'v8::Local<v8::Object> NanNewBufferHandle(char*, size_t, int)':
../node_modules/nan/nan.h:561:50: error: 'callback' was not declared in this scope
         v8::Isolate::GetCurrent(), data, length, callback, hint);
                                                  ^
../node_modules/nan/nan.h:561:60: error: 'hint' was not declared in this scope
         v8::Isolate::GetCurrent(), data, length, callback, hint);
                                                            ^
../node_modules/nan/nan.h: In function 'v8::Local<v8::Object> NanNewBufferHandle(const char*, uint32_t)':
../node_modules/nan/nan.h:568:67: error: call of overloaded 'New(v8::Isolate*, const char*&, uint32_t&)' is ambiguous
     return node::Buffer::New(v8::Isolate::GetCurrent(), data, size);
                                                                   ^
../node_modules/nan/nan.h:568:67: note: candidates are:
In file included from ../node_modules/nan/nan.h:25:0,
                 from ../src/addon.cc:26:
/home/pi/node_modules/raspi-io/node_modules/raspi-pwm/.node-gyp/4.0.0/include/node/node_buffer.h:31:40: note: v8::MaybeLocal<v8::Object> node::Buffer::New(v8::Isolate*, v8::Local<v8::String>, node::encoding) <near match>
 NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                        ^
/home/pi/node_modules/raspi-io/node_modules/raspi-pwm/.node-gyp/4.0.0/include/node/node_buffer.h:31:40: note:   no known conversion for argument 3 from 'uint32_t {aka unsigned int}' to 'node::encoding'
/home/pi/node_modules/raspi-io/node_modules/raspi-pwm/.node-gyp/4.0.0/include/node/node_buffer.h:43:40: note: v8::MaybeLocal<v8::Object> node::Buffer::New(v8::Isolate*, char*, size_t) <near match>
 NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                        ^
/home/pi/node_modules/raspi-io/node_modules/raspi-pwm/.node-gyp/4.0.0/include/node/node_buffer.h:43:40: note:   no known conversion for argument 2 from 'const char*' to 'char*'
In file included from ../src/addon.cc:26:0:
../node_modules/nan/nan.h: In function 'v8::Local<v8::Object> NanNewBufferHandle(uint32_t)':
../node_modules/nan/nan.h:572:61: error: could not convert 'node::Buffer::New(v8::Isolate::GetCurrent(), size)' from 'v8::MaybeLocal<v8::Object>' to 'v8::Local<v8::Object>'
     return node::Buffer::New(v8::Isolate::GetCurrent(), size);
                                                             ^
[... 123 lines cut ...]

[... cut ...]
npm ERR! 
npm ERR! Failed at the raspi-pwm@1.3.3 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the raspi-pwm package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls raspi-pwm
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/npm-debug.log

rudemex, 2015-10-09:
pi@woonkypi ~ $ sudo npm install johnny-five
-
> serialport@1.7.4 install /home/pi/node_modules/johnny-five/node_modules/serialport
> node-pre-gyp install --fallback-to-build

gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/4.0.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/node_modules/johnny-five/node_modules/serialport/.node-gyp"
make: Entering directory '/home/pi/node_modules/johnny-five/node_modules/serialport/build'
make: *** No rule to make target '../.node-gyp/4.0.0/include/node/common.gypi', needed by 'Makefile'.  Stop.
make: Leaving directory '/home/pi/node_modules/johnny-five/node_modules/serialport/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:270:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Linux 4.1.7-v7+
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/home/pi/node_modules/johnny-five/node_modules/serialport/build/serialport/v1.7.4/Release/node-v46-linux-arm/serialport.node" "--module_name=serialport" "--module_path=/home/pi/node_modules/johnny-five/node_modules/serialport/build/serialport/v1.7.4/Release/node-v46-linux-arm"
gyp ERR! cwd /home/pi/node_modules/johnny-five/node_modules/serialport
gyp ERR! node -v v4.0.0
gyp ERR! node-gyp -v v3.0.1
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/node_modules/johnny-five/node_modules/serialport/build/serialport/v1.7.4/Release/node-v46-linux-arm/serialport.node --module_name=serialport --module_path=/home/pi/node_modules/johnny-five/node_modules/serialport/build/serialport/v1.7.4/Release/node-v46-linux-arm' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/pi/node_modules/johnny-five/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:87:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:817:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
node-pre-gyp ERR! System Linux 4.1.7-v7+
node-pre-gyp ERR! command "/usr/local/bin/node" "/home/pi/node_modules/johnny-five/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/pi/node_modules/johnny-five/node_modules/serialport
node-pre-gyp ERR! node -v v4.0.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.7
node-pre-gyp ERR! not ok 
Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/node_modules/johnny-five/node_modules/serialport/build/serialport/v1.7.4/Release/node-v46-linux-arm/serialport.node --module_name=serialport --module_path=/home/pi/node_modules/johnny-five/node_modules/serialport/build/serialport/v1.7.4/Release/node-v46-linux-arm' (1)
npm WARN optional dep failed, continuing serialport@1.7.4
johnny-five@0.8.100 node_modules/johnny-five
├── ease-component@1.0.0
├── descriptor@0.1.0
├── color-convert@0.5.3
├── temporal@0.4.1
├── es6-shim@0.33.6
├── nanotimer@0.3.10
├── chalk@1.1.1 (escape-string-regexp@1.0.3, supports-color@2.0.0, ansi-styles@2.1.0, has-ansi@2.0.0, strip-ansi@3.0.0)
├── lodash@3.10.1
├── array-includes@3.0.1 (define-properties@1.1.1, es-abstract@1.3.2)
└── firmata@0.5.5 (object-assign@1.0.0, browser-serialport@2.0.3, es6-map@0.1.1)

nebrius, 2015-10-09:
Ouch. I'm going to go out on a limb and say this is because raspi-io still uses NAN1. Once raspi-io is upgraded to NAN2, I _think_ this will be fixed. Fingers crossed 😅

rudemex, 2015-10-09:
shit! jajajaja i need j5 on my raspi XD

nebrius, 2015-10-09:
Do you need to use jessie? Wheezy is still available for download and should work fine.

fivdi, 2015-10-09:
Alternatively, use Node.js 0.12 on Jessie

nebrius, 2015-10-09:
Sorry, that was my bad, I read ahead too quickly and didn't see that you were using node 4. @fivdi is right, just downgrading to 0.12 should work.

For the record, J5 also doesn't work on node 4 for the same reason. The upgrade process is being tracked over at https://github.com/voodootikigod/node-serialport/issues/578

rudemex, 2015-10-09:
ok, thanks @fivdi and @nebrius I'll try this solution.

fivdi, 2015-10-09:
serialport is an optional dependency in J5. It's not always needed. In particular, it's not needed for Raspi-IO. The serialport compile errors displayed when J5 is installed can be ignored in this case.

nebrius, 2015-10-09:
oh cool, good to know @fivdi. For some reason I thought it was required. As long as you don't need to access an Arduino, that won't be a blocker.

fivdi, 2015-10-09:
So ist es :)

rudemex, 2015-10-10:
great, i will try this solutions now, thanks @fivdi @nebrius

rudemex, 2015-10-10:
is working.

Thanks @fivdi @nebrius .

closed 2015-10-10 by rudemex

nebrius, 2015-10-10:
👍