rwaldron/johnny-five #922: Can We connect from host to arduino by wifi or bluetooth.

jackyphuong, 2015-10-11:


rwaldron, 2015-10-11:
cc @monteslu @ajfisher

ajfisher, 2015-10-12:
@jackyphuong are you simply trying to talk to you arduino over wifi / bluetooth or are you trying to flash it that way?

If it's simply replace the cable for talking to it (once firmata is already on the board) then there's two paths for you:

Bluetooth: 

Depends on your module as to whether you're using HC-05 or HC06 (and they are different)

HC-05: https://github.com/rwaldron/johnny-five/wiki/Getting-Started-with-Johnny-Five-and-HC-05-Bluetooth-Serial-Port-Module
HC-06: https://github.com/rwaldron/johnny-five/wiki/Getting-Started-with-Johnny-Five-and-JY-MCU-Bluetooth-Serial-Port-Module

Current method of getting wifi going using ESP8266 using UDP-Serial Bridge
https://gist.github.com/ajfisher/5fe60fe7d8c49b3223f0

jackyphuong, 2015-10-16:
Thanks @ajfisher, it really only replace cable. I followed link use bluetooth HC-06, it works well

closed 2015-10-16 by jackyphuong

ajfisher, 2015-10-16:
Excellent - good to hear!