rwaldron/johnny-five #925: Neopixel/RGB LED strips

alexcroox, 2015-10-14:
Does Johnny-five support Adafruit Neopixels or similar RGB LED strips?

frxnz, 2015-10-14:
You can use Neopixels with Johnny-Five using [node-pixel](https://github.com/ajfisher/node-pixel). Node-pixel supports WS2812, and I *believe* WS2811 strips will work as well.

rwaldron, 2015-10-14:
Thanks @frxnz

closed 2015-10-14 by rwaldron

alexcroox, 2015-10-14:
Ah perfect thank you. And for the [non addressable strips](https://www.sparkfun.com/products/12023) what would you recommend?

alexcroox, 2015-10-15:
https://learn.adafruit.com/rgb-led-strips/usage are the non addressable wiring diagram. 

I'm guessing I could use J5 LED RGB? https://github.com/rwaldron/johnny-five/blob/master/docs/led-rgb.md

frxnz, 2015-10-15:
Yep. Most non-addressable strips should just work with the `Led`/`Led.RGB` classes.

alexcroox, 2015-10-15:
Appreciate it thank you!