rwaldron/johnny-five #926: Improved support for motion sensor

Anfa-am, 2015-10-15:
There is ok support for motion sensors but the majority of the time it is sporadic, unpredictable and unreliable.

dtex, 2015-10-15:
Can you give us more info? Which motion sensor(s), what you are seeing, and what you expect to see.

Anfa-am, 2015-10-15:
IR Infrared Motion Detection Sensor Module (DC 5V~20V)

Its an io device so I'm expecting to see a 1 or a 0. problem is it only returns accurately about 1/3 of the time.

rwaldron, 2015-10-15:
@Anfa-am those devices have potentiometers on them to adjust sensitivity and reading cycle time:

[image]

Johnny-Five's `Motion` class is a very minimal OO abstraction that just processes whatever signal it receives into events. I've found that the cheaper the source of these devices, the lower the quality and accuracy. For example: the Parallax version of this sensor is really accurate, but [these](http://www.amazon.com/HC-SR501-Adjust-Pyroelectric-Infrared-Detector/dp/B00QQ0NIH0) are not all equally accurate (maybe half are usable).

Anfa-am, 2015-10-15:
@rwaldron, I see. Going through the j5 source for motion it became obvious that its not the issue as its mostly callbacks from the motion sensor's return. 

I had originally thought it was a library issue because it works perfectly in C. Perhaps calibrating the module will help eliminate some of the variable inconstancy.

Regardless of the solution it is safe to assume that it is not a software one.

rwaldron, 2015-10-17:
> I had originally thought it was a library issue because it works perfectly in C.

Can you share the specific code that you're testing with?

rwaldron, 2015-11-05:
@anfa-am I'm going to reclose this, but you can reopen if you want to dig deeper in the future. I'm still interested in the C code that you used. Be sure to @-mention me if you ever come back to this.

closed 2015-11-05 by rwaldron