rwaldron/johnny-five #927: Is Mux Shield II compatible with J5?

aabm00, 2015-10-16:
Hello

I need to expand my Arduino Yun  with more Analog imputs. I have found this shield "Mux Shield II". 
https://www.sparkfun.com/products/11723
It comes with a library In arduino languaje that has to be included in the sketch to be executed in the board.

I'm using node/J5, so it's supposed that this library would have to be implemented in Javascript to work, isn't it?

I have been looking in the API and haven't seen any shiels of this kind. 

Is Mux Shield II compatible with J5?
Are there any shield to expand Digital and analogics I/O compatible with J5?

Thanks

rwaldron, 2015-10-17:
Looking at the cpp library code, this could easily be supported as an [`Expander`](http://johnny-five.io/api/expander/) component. I've just ordered one, once I receive it we can start working on a controller for it.

aabm00, 2015-10-20:
Thanks. 
How shall I know when the controlled will be available?

For your answer I suppose that any component needs a javascript controller to be operative with J5. Am I right?

rwaldron, 2015-10-20:
> For your answer I suppose that any component needs a javascript controller to be operative with J5. Am I right?

Yep!

I just got the thing assembled today, so I will work on this over the next few days :)

closed 2015-11-02 by rwaldron