rwaldron/johnny-five #929: What's required to support node 4

aredridel, 2015-10-18:
I just managed to hack things together, but there's a whole chain of dependency upgrades needed.

`serialport@2.0.1` is out, supporting nan 2 and therefore node 4. However, it doesn't support node 0.10. I've created `metaserialport` that requires both as optional dependencies (and so only failing if both fail to install).

`firmata` will have to be updated to use `serialport@2.0.1` -- work is in progress, though, and I've made a PR (https://github.com/jgautier/firmata/pull/101) to that, and then `johnny-five` will have to be updated to support a new `firmata`.

aredridel, 2015-10-18:
A pull request to use metaserialport is at https://github.com/rwaldron/johnny-five/pull/930

rwaldron, 2015-10-19:
What's required: time to finish testing a version specific node-serialport install routine. It's in progress and I'm not rushing it.

aredridel, 2015-10-19:
That too!

dtex, 2015-10-19:
@rwaldron is there an open issue/pr for the version specific install routine? Is that an npm thing or a node-serialport thing? I liked your idea for package.json.

rwaldron, 2015-10-19:
>  I liked your idea for package.json.

Me too, but it's not going to happen :( I think @aredridel's solution looks close to that, but I am only just now getting a chance to sit down and test it. If it does the deps handling the way I want it to, then that's the path for Johnny-Five

fivdi, 2015-10-19:
> `serialport@2.0.1` is out, supporting nan 2 and therefore node 4. However, it doesn't support node 0.10. 

@aredridel is the lack of support for node 0.10 in certain use cases or is it a general issue? I tried the original PR for NAN2 support with node 0.10.38 on a BeagleBone and what I tried worked.

rwaldron, 2015-10-19:
I've written an explanation and updated firmata.js's postinstall script. I've also copied the same script here (as a work in progress). Please test with `npm install git://github.com/rwaldron/johnny-five.git`

rwaldron, 2015-10-19:
Current results from installing and running a program: 

```
rwaldron at nova in ~/robotz/j5-install-test
$ ls
.
└── [-rw-r--r-- rwaldron staff     202 Oct 19 16:08]  blink.js

0 directories, 1 file
rwaldron at nova in ~/robotz/j5-install-test
$ cat blink.js
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {

  var led = new five.Led(13);

  // "blink" the led in 500ms on-off phase periods
  led.blink(500);
});
rwaldron at nova in ~/robotz/j5-install-test
$ node -v
v4.2.1
rwaldron at nova in ~/robotz/j5-install-test
$ npm install git://github.com/rwaldron/johnny-five.git
\
> firmata@0.7.1-pre-2 postinstall /Users/rwaldron/robotz/j5-install-test/node_modules/johnny-five/node_modules/firmata
> node scripts/postinstall

\
> serialport@2.0.1 install /Users/rwaldron/robotz/j5-install-test/node_modules/johnny-five/node_modules/firmata/node_modules/serialport
> node-pre-gyp install --fallback-to-build

  CXX(target) Release/obj.target/serialport/src/serialport.o

  CXX(target) Release/obj.target/serialport/src/serialport_unix.o

  CXX(target) Release/obj.target/serialport/src/serialport_poller.o

  SOLINK_MODULE(target) Release/serialport.node

  SOLINK_MODULE(target) Release/serialport.node: Finished

  COPY /Users/rwaldron/robotz/j5-install-test/node_modules/johnny-five/node_modules/firmata/node_modules/serialport/build/serialport/v2.0.1/Release/node-v46-darwin-x64/serialport.node

  TOUCH Release/obj.target/action_after_build.stamp

serialport@2.0.1 node_modules/serialport
├── bindings@1.2.1
├── async@0.9.0
├── sf@0.1.7
├── nan@2.0.9
├── debug@2.2.0 (ms@0.7.1)
└── optimist@0.6.1 (wordwrap@0.0.3, minimist@0.0.10)

> johnny-five@0.8.106 postinstall /Users/rwaldron/robotz/j5-install-test/node_modules/johnny-five
> node scripts/postinstall

johnny-five@0.8.106 node_modules/johnny-five
├── ease-component@1.0.0
├── descriptor@0.1.0
├── color-convert@0.5.3
├── temporal@0.4.1
├── semver@5.0.3
├── nanotimer@0.3.10
├── es6-shim@0.33.6
├── chalk@1.1.1 (escape-string-regexp@1.0.3, supports-color@2.0.0, ansi-styles@2.1.0, strip-ansi@3.0.0, has-ansi@2.0.0)
├── mv@2.1.1 (ncp@2.0.0, mkdirp@0.5.1, rimraf@2.4.3)
├── lodash@3.10.1
├── array-includes@3.0.1 (define-properties@1.1.2, es-abstract@1.4.0)
└── firmata@0.7.1-pre-2 (browser-serialport@2.0.3)
rwaldron at nova in ~/robotz/j5-install-test
$ node blink.js
1445285485766 Device(s) /dev/cu.usbmodem1421
1445285485775 Connected /dev/cu.usbmodem1421
1445285489242 Repl Initialized
>> I can see it working.
>>
(^C again to quit)
>>
1445285501658 Board Closing.
rwaldron at nova in ~/robotz/j5-install-test
$ nvm use 0.12.4
Now using node v0.12.4
rwaldron at nova in ~/robotz/j5-install-test
$ node -v
v0.12.4
rwaldron at nova in ~/robotz/j5-install-test
$ rm -r node_modules/
rwaldron at nova in ~/robotz/j5-install-test
$ npm install git://github.com/rwaldron/johnny-five.git

> firmata@0.7.1-pre-2 postinstall /Users/rwaldron/robotz/j5-install-test/node_modules/johnny-five/node_modules/firmata
> node scripts/postinstall

\
> serialport@1.7.4 install /Users/rwaldron/robotz/j5-install-test/node_modules/johnny-five/node_modules/firmata/node_modules/serialport
> node-pre-gyp install --fallback-to-build

[serialport] Success: "/Users/rwaldron/robotz/j5-install-test/node_modules/johnny-five/node_modules/firmata/node_modules/serialport/build/serialport/v1.7.4/Release/node-v14-darwin-x64/serialport.node" is installed via remote

serialport@1.7.4 node_modules/serialport
├── bindings@1.2.1
├── sf@0.1.7
├── async@0.9.0
├── nan@1.8.4
├── debug@2.2.0 (ms@0.7.1)
└── optimist@0.6.1 (wordwrap@0.0.3, minimist@0.0.10)

> johnny-five@0.8.106 postinstall /Users/rwaldron/robotz/j5-install-test/node_modules/johnny-five
> node scripts/postinstall

johnny-five@0.8.106 node_modules/johnny-five
├── ease-component@1.0.0
├── descriptor@0.1.0
├── color-convert@0.5.3
├── temporal@0.4.1
├── semver@5.0.3
├── chalk@1.1.1 (escape-string-regexp@1.0.3, supports-color@2.0.0, ansi-styles@2.1.0, has-ansi@2.0.0, strip-ansi@3.0.0)
├── es6-shim@0.33.6
├── nanotimer@0.3.10
├── mv@2.1.1 (ncp@2.0.0, mkdirp@0.5.1, rimraf@2.4.3)
├── array-includes@3.0.1 (define-properties@1.1.2, es-abstract@1.4.0)
├── lodash@3.10.1
└── firmata@0.7.1-pre-2 (browser-serialport@2.0.3)
rwaldron at nova in ~/robotz/j5-install-test
$ node blink.js
1445285544313 Device(s) /dev/cu.usbmodem1421
1445285544330 Connected /dev/cu.usbmodem1421
1445285547797 Repl Initialized
>> This also works!
>>
(^C again to quit)
>>
1445285557616 Board Closing.
```

aredridel, 2015-10-19:
Oh, interesting. serialport-2 didn't work for me at all on node 0.10.40 on my mac.

rwaldron, 2015-10-25:
> Oh, interesting. serialport@2 didn't work for me at all on node 0.10.40 on my mac.

Is this still the case?

aredridel, 2015-10-27:
It is not!

closed 2015-10-27 by aredridel

rwaldron, 2015-10-27:
Thanks for the follow up!

aredridel, 2015-10-27:
Sure thing!