rwaldron/johnny-five #932: Hi, getting this error on install, any idea on the update for serialports for node 4.x.x?:

veebuv, 2015-10-19:
^
9 errors generated.
make: *** [Release/obj.target/serialport/src/serialport.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:270:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Darwin 15.0.0
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/Users/VaibhavNamburi/Desktop/Udemy/arduino/node_modules/johnny-five/node_modules/serialport/build/serialport/v1.7.4/Release/node-v46-darwin-x64/serialport.node" "--module_name=serialport" "--module_path=/Users/VaibhavNamburi/Desktop/Udemy/arduino/node_modules/johnny-five/node_modules/serialport/build/serialport/v1.7.4/Release/node-v46-darwin-x64"
gyp ERR! cwd /Users/VaibhavNamburi/Desktop/Udemy/arduino/node_modules/johnny-five/node_modules/serialport
gyp ERR! node -v v4.1.2
gyp ERR! node-gyp -v v3.0.3
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/Users/VaibhavNamburi/Desktop/Udemy/arduino/node_modules/johnny-five/node_modules/serialport/build/serialport/v1.7.4/Release/node-v46-darwin-x64/serialport.node --module_name=serialport --module_path=/Users/VaibhavNamburi/Desktop/Udemy/arduino/node_modules/johnny-five/node_modules/serialport/build/serialport/v1.7.4/Release/node-v46-darwin-x64' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/Users/VaibhavNamburi/Desktop/Udemy/arduino/node_modules/johnny-five/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:87:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:818:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
node-pre-gyp ERR! System Darwin 15.0.0
node-pre-gyp ERR! command "/usr/local/bin/node" "/Users/VaibhavNamburi/Desktop/Udemy/arduino/node_modules/johnny-five/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /Users/VaibhavNamburi/Desktop/Udemy/arduino/node_modules/johnny-five/node_modules/serialport
node-pre-gyp ERR! node -v v4.1.2
node-pre-gyp ERR! node-pre-gyp -v v0.6.7
node-pre-gyp ERR! not ok 
Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/Users/VaibhavNamburi/Desktop/Udemy/arduino/node_modules/johnny-five/node_modules/serialport/build/serialport/v1.7.4/Release/node-v46-darwin-x64/serialport.node --module_name=serialport --module_path=/Users/VaibhavNamburi/Desktop/Udemy/arduino/node_modules/johnny-five/node_modules/serialport/build/serialport/v1.7.4/Release/node-v46-darwin-x64' (1)
npm WARN optional dep failed, continuing serialport@1.7.4

dtex, 2015-10-19:
serialport is getting there, and they have a release for node 4.x but I think they are still ironing out issues on some platforms:

https://github.com/voodootikigod/node-serialport/issues/578

We can't just change the dependency here because it would break things for those cases and more importantly for anyone NOT running node 4.x. I see in another issue that a fix to allow different seriaport installs depending on node version is in the works.

veebuv, 2015-10-19:
Hey dtex, fixed it. Didn't want to rollback to 0.12.x so i just installed 'n' and switch between versions, this made life easy. serialport is supported on that version. I'll permanently switch back when the issue is fixed on this side, and from the looks of the previous issues, it's being worked on

rwaldron, 2015-10-25:
Please update to latest Johnny-Five!

closed 2015-10-25 by rwaldron