rwaldron/johnny-five #933: node-red timeout

barabasj, 2015-10-19:
I'm experiencing the same problem described in #767. Trying to start node-red with Johnny-five enabled, it starts and ends with a timeout error. Running node samples it works just fine.

Any idea?
Regards, Javier.

rwaldron, 2015-10-25:
Can you share the exact error that you're getting? Thanks!

barabasj, 2015-10-25:
Rick,
Thanks for your answer. The complete starting screen follows:

[image]
In text:
Welcome to Node-RED
===================

25 Oct 12:08:47 - [info] Node-RED version: v0.11.2
25 Oct 12:08:47 - [info] Node.js  version: v0.10.36
25 Oct 12:08:47 - [info] Loading palette nodes
25 Oct 12:08:48 - [warn] ------------------------------------------
25 Oct 12:08:48 - [warn] Failed to register 1 node type
25 Oct 12:08:48 - [warn] Run with -v for details
25 Oct 12:08:48 - [warn] ------------------------------------------
25 Oct 12:08:48 - [info] Settings file  : /home/barabasj/.node-red/settings.js
25 Oct 12:08:48 - [info] User directory : /home/barabasj/.node-red
25 Oct 12:08:48 - [info] Flows file : /home/barabasj/.node-red/flows_oc7331236382.ibm.com.json
25 Oct 12:08:48 - [info] Server now running at http://127.0.0.1:1880/
25 Oct 12:08:48 - [info] Starting flows
25 Oct 12:08:48 - [info] Started flows
1445785737197 Device or Firmware Error A timeout occurred while connecting to the Board. 

Please check that you've properly flashed the board with the correct firmware.
See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting  
25 Oct 12:08:57 - [red] Uncaught Exception:
25 Oct 12:08:57 - TypeError: Uncaught, unspecified "error" event.
    at TypeError (<anonymous>)
    at Board.emit (events.js:74:15)
    at Board.log (/home/barabasj/.node-red/node_modules/johnny-five/lib/board.js:629:8)
    at Board.(anonymous function) [as error] (/home/barabasj/.node-red/node_modules/johnny-five/lib/board.js:640:14)
    at Board.<anonymous> (/home/barabasj/.node-red/node_modules/johnny-five/lib/board.js:411:14)
    at Timer.listOnTimeout [as ontimeout] (timers.js:112:15)
[barabasj@oc7331236382 .node-red]$ 

Regards!
Javier.

barabasj, 2015-10-25:
My questions still open. I closed it by error.

rwaldron, 2015-11-05:
Sorry I missed your reply! Where is this code running, i.e. on a raspberry pi (like the other issue you mentioned)?

barabasj, 2015-11-05:
No. In my case it is running in a Arduino Mega. If I run the code from the
command line it works. From Node-Red fails.
Regards, Javier.

claudiuo, 2016-02-07:
Sorry to chime in but I am having the same problem. I just installed johnny-five and monteslu's nod-red-contrib-gpio on my Linux Mint 17 machine using latest node-red (0.13.0). Configured johnny-five to use firmata and created a led and added led.blink. The interesting thing is that it worked once. Then I changed the code a bit and every time I deploy I get a similar error as the OP (please see details below).

If I try johnny-five directly on command line it works great. I tried to restart the machine but same error. The only thing I can do is delete the flows file to be able to start node-red.

Any idea what else I could try? Thanks in advance for any help.

   claudiu

Error details:
```454883661393 Device or Firmware Error A timeout occurred while connecting to the Board. 

Please check that you've properly flashed the board with the correct firmware.
See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting 
7 Feb 15:21:01 - [red] Uncaught Exception:
7 Feb 15:21:01 - Error: A timeout occurred while connecting to the Board.
    at Board.<anonymous> (/home/tavi/.node-red/node_modules/johnny-five/lib/board.js:405:26)
    at Timer.listOnTimeout [as ontimeout] (timers.js:110:15)```

marvelousNinja, 2016-03-21:
Hi there, I might have a solution.
In my case, there was some weird conflict between
`node-red-node-arduino` and `node-red-contrib-gpio` (which uses Johny-Five). Here's what I did:
* Updated Node to 5.9 (didn't help)
* Removed `node-red-arduino` package (still nothing)
* Removed all existing Node RED configs and stuff from `~/.node-red` (bingo!)

Hope this helps!

reconbot, 2016-05-25:
@marvelousNinja that's great! Thank you.

@barabasj I'm going to close this due to age, but if @marvelousNinja didn't solve it for you feel free to reopen!

closed 2016-05-25 by reconbot