rwaldron/johnny-five #936: Adafruit Audio FX Sound Board Support

topkoa, 2015-10-22:
Hey I've ordered a couple of these for my testing and would like to use it in an up-coming project...

Looking for some advice on connecting it to an Arduino/J5, specifically should I control it via the Serial/UART or via GPIO pins? (It can handle both, so I'm looking for the (easiest) most accepted solution when controlling it via JS/J5)

Thanks!

https://learn.adafruit.com/adafruit-audio-fx-sound-board/triggering-audio

rwaldron, 2015-10-23:
> Looking for some advice on connecting it to an Arduino/J5, specifically should I control it via the Serial/UART or via GPIO pins

Support is inbound! https://github.com/jgautier/firmata/pull/97

topkoa, 2015-10-23:
Awesome!

Looking forward to seeing it land!

rwaldron, 2015-10-25:
I'll keep this open until we can come back to it :)

topkoa, 2015-10-29:
Hey I see support landed over with firmata! 

I'm looking forward to checking it out.  I have a question, however-- can the ".serialConfig" (et al) be utilized within the confines of the J5 API? Or is that support still on it's way?

dtex, 2016-01-23:
@topkoa All the firmata serial commands are exposed on the board.io object (which by default is an instance of firmata) so:

````
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
  this.io.serialConfig({ portId: 8, baud: 9600, rxPin: 10, txPin: 11 });
  this.io.serialRead(8, function(data) {
    console.log(data);
  });
});
````

The constants for portID's are defined in firmata. I just used "8" which is SW_SERIAL0.

dtex, 2018-01-11:
Hi @topkoa  

Johnny-Five contributors and maintainers are loathe to close issues where someone has a need. We don't want anyone to ever feel that we don't care, but we just haven't gotten to this feature yet. Rather than leave it languishing as an open issue we have created a [Requested Features](../wiki/Requested-Features) page and added this devices request under the new class "Sound Generator/Synthesizer" listed there.

closed 2018-01-11 by dtex