rwaldron/johnny-five #938: led animations keep playing when another method (i.e. blink) is called

dtex, 2015-10-23:
Running this:
````
led.pulse();
led.blink();
````
gives a weird, twitchy pulse. We could stop animations when certain methods are called, but is that desired behavior?

rwaldron, 2015-10-23:
IIRC, that used to just stop the pulse and do a blink, it would happen fast enough that pulse would likely never be seen. Any "cyclic" thing should kill a previous/currently running "cyclic" thing.

rwaldron, 2015-10-25:
@dtex ok, so I was right above... We needed better tests for this change: https://github.com/rwaldron/johnny-five/commit/d12f8fba241c6d1e074cffa7e030762f9e12f400#diff-cef4895508501dd891236f06ba4cadfe Can you just call `state.animation.stop()` before this line: https://github.com/rwaldron/johnny-five/commit/d12f8fba241c6d1e074cffa7e030762f9e12f400#diff-cef4895508501dd891236f06ba4cadfeR384 ?

dtex, 2015-10-26:
I'll make that change this evening. There are a couple of other places I need to add it but not many (pulse, fade, servo.sweep). Also, I can't forget about refactoring animation so that it doesn't use temporal for more than a couple of seconds.

rwaldron, 2015-11-05:
@dtex can you write up a patch for this?

dtex, 2015-11-05:
Rodger

dtex, 2015-11-06:
Closed with #954

closed 2015-11-06 by dtex