rwaldron/johnny-five #941: Improve usability of uploading PingFirmata

scottgonzalez, 2015-10-26:
Copying the source for PingFirmata isn't very friendly right now. We can improve this by linking to just the source (enabling select all, copy; or save as) and by providing a button to copy the content to your clipboard (via clipboard.js). The clipboard part might need to be specific to johnny-five.io.

rwaldron, 2015-10-26:
I don't like the idea of further encouraging people to use PingFirmata. It needs to DIAF. There are many alternatives that don't require a potentially hazardous amount of  process blocking IO.

scottgonzalez, 2015-10-26:
I don't think this change would increase the encouragement to use PingFirmata, it would just make it easier for those that *need* to use it. If the components that currently require PingFirmata can be changed to not require it, that'd be ideal. But as long as it's required for some components, it should probably be easier to upload.

soundanalogous, 2015-10-26:
I'll add a [NewPing](http://playground.arduino.cc/Code/NewPing) wrapper to ConfigurableFirmata at some point. NewPing is timer-based and thus non-blocking... but it will affect other features that depend on timers (such as PWM, Servos, etc) on certain pins depending on the board in use. @MrYsLab had an initial implementation of a NewPing wrapper at one point.

soundanalogous, 2015-10-26:
The i2c backpack approach that @ajfisher is taking is likely the best solution however since it won't mess up timer related functionality on other pins.

ajfisher, 2015-10-26:
So pretty much got a working version of this in node-interchange as this is my test case. Interchange will install either PingFirmata directly on the arduino or the I2C backpack version depending on what you tell it to do. That will at least make the installation as easy as:

```
interchange install PingFirmata -a uno -p /dev/tty.usbserial1432
```

Though I agree with @rwaldron PingFirmata isn't wonderful so the backpack is the better option in that it won't affect other devices connected - and it's rare you want to use a ping sensor without something to react to it.

This is all working in interchange but needs some testing (esp on windows) still and a couple of final syntactic sugar bits in interchange and then I'll release to npm - but you can install using git and it should work. Am about 2 evenings from having this part sorted I reckon.

ajfisher, 2015-12-15:
So this is now all working in node interchange so I'd think this is largely closed.

For reference command would be:

```
interchange install hcsr04 -a uno -p /dev/tty.usbserial1432 --firmata
```

And that would do the job.

rwaldron, 2015-12-15:
@ajfisher can you add this to the bottom of https://github.com/rwaldron/johnny-five/wiki/Proximity#pingfirmata ?

ajfisher, 2015-12-15:
Yep - should be able to get that in today

ajfisher, 2015-12-15:
Check https://github.com/rwaldron/johnny-five/wiki/Proximity for updates on this front across the board, including documentation around this.

As I've now rolled pingFirmata into the interchange HCSR04 stuff is it worth even removing the reference to that raw pingFirmata file of yours @rwaldron now?

dtex, 2016-01-23:
It's not clear to me what needs to happen here. Should we add disclaimers everywhere pingfirmata is still used (docs/proximity-hcsr04.md) and then close this issue?

ajfisher, 2016-01-23:
I thought I'd caught it where I could see it...

ajfisher, 2016-02-16:
Suggest this is closed now?

scottgonzalez, 2016-02-16:
Yes, the updates look good. Thanks.

closed 2016-02-16 by scottgonzalez