rwaldron/johnny-five #944: Creating new Board with SerialPort

jyepesr1, 2015-10-29:
I've tried running this code, but always I have the same error

```javascript
var SerialPort = require("serialport").SerialPort;
var five = require("johnny-five")
var board = new five.Board({
   port: new SerialPort("/dev/ttyACM0", {
      baudrate: 9600
   })
});

board.on("ready", function() {
   var temperature = new five.Temperature({
      controller: "LM35",
      pin: "A0",
      freq: 4000
   });

   temperature.on("data", function() {
      celsiustemp = Number(this.celsius.toFixed(2));
      console.log(celsiustemp + " °C");
      /*serialPort.open(function (error) {
         if ( error ) {
            console.log('failed to open: '+error);
         } else {
            console.log('open');
            setInterval(function(){
               var sending=8+"\n"
               serialPort.write(sending);
               console.log(sending);
            }, 1000);
         }
      });*/
   });
});

```

1446126080144 SerialPort /dev/ttyACM0  
1446126080145 Connected /dev/ttyACM0  
1446126090151 Device or Firmware Error A timeout occurred while connecting to the Board. 

Please check that you've properly flashed the board with the correct firmware.
See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting  

TypeError: Uncaught, unspecified "error" event.
    at TypeError (<anonymous>)
    at Board.emit (events.js:74:15)
    at Board.log (/home/johan/Desktop/Git/johnny-five/temperatureSensor/node_modules/johnny-five/lib/board.js:629:8)
    at Board.(anonymous function) [as error] (/home/johan/Desktop/Git/johnny-five/temperatureSensor/node_modules/johnny-five/lib/board.js:640:14)
    at Board.<anonymous> (/home/johan/Desktop/Git/johnny-five/temperatureSensor/node_modules/johnny-five/lib/board.js:411:14)
    at Timer.listOnTimeout [as ontimeout] (timers.js:121:15)

I'm new in johnny-five. Can anybody help me?
Thanks.

rwaldron, 2015-10-29:
Are you connecting to a board that has StandardFirmata running on it? If yes, then you can't change te baudrate to 9600, because StandardFirmata is set to 57600

rwaldron, 2015-10-29:
Just do this: 

```js
var SerialPort = require("serialport").SerialPort;
var five = require("johnny-five")
var board = new five.Board({
   port: "/dev/ttyACM0"
});

board.on("ready", function() {
   var temperature = new five.Temperature({
      controller: "LM35",
      pin: "A0",
      freq: 4000
   });

   temperature.on("data", function() {
      celsiustemp = Number(this.celsius.toFixed(2));
      console.log(celsiustemp + " °C");
   });
});
```

rwaldron, 2016-01-22:
Please re-open if this is still an issue

closed 2016-01-22 by rwaldron