rwaldron/johnny-five #948: Reading temperature and pressure stops after few seconds

mjuanfe, 2015-10-31:
I'm trying this example:

		var five = require("johnny-five");
		var board = new five.Board();

		board.on("ready", function() {
		  var multi = new five.Multi({
		    controller: "BMP180"
		  });

		  multi.on("change", function() {
		    console.log("temperature");
		    console.log("  celsius      : ", this.temperature.celsius);
		    console.log("  fahrenheit   : ", this.temperature.fahrenheit);
		    console.log("  kelvin       : ", this.temperature.kelvin);
		    console.log("--------------------------------------");

		    console.log("barometer");
		    console.log("  pressure     : ", this.barometer.pressure);
		    console.log("--------------------------------------");
		  });
		});

After run it works but after few seconds stops giving the information, why this happen? I'm doing something wrong? 

After few seconds windows prompt is again on my route:

			--------------------------------------
			temperature
			  celsius      :  21.6
			  fahrenheit   :  70.88
			  kelvin       :  294.75
			--------------------------------------
			barometer
			  pressure     :  102.0220
			--------------------------------------
			temperature
			  celsius      :  21.6
			  fahrenheit   :  70.88
			  kelvin       :  294.75
			--------------------------------------
			barometer
			  pressure     :  102.0180
			--------------------------------------

			C:\nodes\arduino>

fivdi, 2015-10-31:
If Node.js v4 or higher is being then the issue is highly likely to be related to serialport which has issues with Node.js 4 on Windows. See https://github.com/voodootikigod/node-serialport/issues/578.

mjuanfe, 2015-10-31:
Thanks for the fast response, tomorrow I'll try with a different version of node and I'll tell you how was.

fivdi, 2015-10-31:
AFAIK everything should be fine with Node.js v0.12.x on Windows.

mjuanfe, 2015-11-01:
I installed  nvm for windows and tried the 0.12.2 version, still not working :(

fivdi, 2015-11-01:
Hmm, sorry to hear that. I'm afraid I don't have a windows machine and can't really provide any further information. Is exactly the same thing happening?

mjuanfe, 2015-11-01:
Yes, the same result, I think I'll try to install a VM with Ubuntu, anyway bothers me not solve it! grrr...

dtex, 2018-01-16:
Hi @mjuanfe Since we haven't heard back I'm going to close this. Please feel free to re-open this any time if it is still an issue for you.

closed 2018-01-16 by dtex

mjuanfe, 2018-01-16:
Hi @dtex , thanks for update this old issue, I think it's solved.