rwaldron/johnny-five #955: Sensor: Boolean barrier should be based on raw value

scottgonzalez, 2015-11-10:
The boolean barrier is currently based on the scaled value, but it seems like it should be based on the raw value.

If there's consensus on this, I can put together a PR.

rwaldron, 2015-11-10:
I think this would give strange results: 

```js
var scale = [0, 100];
var adjuster = new five.Sensor({ pin: "A0", scale: scale });
var input = new five.Sensor({ pin: "A1", scale: scale });
var output = new five.Led(13);

// I have a potentiometer used to adjust the point in which 
// the input should be reacted on:
adjuster.on("change", function() {
  input.booleanAt(this.value);
});

input.on("change", function() {
  if (this.boolean && !led.isOn) {
    led.on();
  } else {
    if (led.isOn) {
      led.off();
    }
  }
});
```

If the `booleanBarrier` was 0-1023, this program would only sort of work correctly at the very bottom end of the potentiometer's range. If you need to work with the raw voltage reading, then I suggest using the standalone `scale` function, which is `five.Fn.scale(value, fromLow, fromHigh, toLow, toHigh)`.

scottgonzalez, 2015-11-10:
Here's an even simpler case that I had which made me find this:

```js
sensor.scale(0, 100).on("change", function() {
	console.log("change", this.value, this.boolean);
});
```

`this.boolean` is always `false` because the values are scaled.

If this is intended, that's fine. The docs just weren't clear, so I can close this and update the docs if that's the preferred way to handle this.

rwaldron, 2015-11-10:
Maybe this is a documentation issue? In the docs, it says "defaults to 512": http://johnny-five.io/api/sensor/#api maybe it should be clearer somehow?

scottgonzalez, 2015-11-10:
That seems good. I can send a PR for that and update the docs once the PR is merged.

rwaldron, 2015-11-10:
@scottgonzalez good to go for the docs update

closed 2015-11-10 by rwaldron

scottgonzalez, 2015-11-10:
Docs are updated.

PR #960 (2015-11-10): Sensor: Default to 50% of scaled values for boolean barrier
Fixes #955
commit d77b5daf (2015-11-10): Sensor: Default to 50% of scaled values for boolean barrier

Fixes #955