rwaldron/johnny-five #956: Implement HC-SR04-I2C-BACKPACK controller

ajfisher, 2015-11-10:
Implement a controller which works for the HC-SR04 module that works with the backpack at https://github.com/ajfisher/nodebots-hcsr04

@ajfisher to implement - this here as a note to finish it off.

ajfisher, 2015-11-11:
Just got to fix a test - but this is failing on the within test.

```javascript
exports["Proximity: HCSR04I2C"] = {
  setUp: function(done) {
    this.board = newBoard();
    this.clock = sinon.useFakeTimers();
    this.i2cConfig = sinon.spy(MockFirmata.prototype, "i2cConfig");
    this.i2cReadOnce = sinon.stub(MockFirmata.prototype, "i2cReadOnce", function(ADDRESS, BYTES_TO_READ, callback) {
      var pulseval = 1000;
      // Split to HIGH and LOW
      console.log(pulseval);
      callback(pulseval);
    });

    this.proximity = new Proximity({
      controller: "HCSR04I2C",
      freq: 100,
    });

    done();
  },

  tearDown: function(done) {
    this.i2cConfig.restore();
    this.i2cReadOnce.restore();
    this.clock.restore();
    done();
  },

  shape: function(test) {
    test.expect(proto.length + instance.length);

    proto.forEach(function(method) {
      test.equal(typeof this.proximity[method.name], "function");
    }, this);

    instance.forEach(function(property) {
      test.notEqual(typeof this.proximity[property.name], 0);
    }, this);

    test.done();
  },

  data: function(test) {
    var spy = sinon.spy();
    test.expect(1);

    this.proximity.on("data", spy);
    this.clock.tick(100);
    test.equal(spy.callCount, 1);
    test.done();
  },

  change: function(test) {
    test.expect(1);

    var spy = sinon.spy();

    this.proximity.on("change", spy);

    this.clock.tick(100);

    test.ok(spy.called);
    test.done();
  },

    // TODO FIX THIS
  within: function(test) {
    var spy = sinon.spy();
    test.expect(2);

    this.clock.tick(250);

    this.proximity.within([0, 120], "inches", function() {
      test.equal(this.inches, 5.85);
      spy();
    });

    this.clock.tick(100);
    test.ok(spy.calledOnce);
    test.done();
  }
};
```

Error:

```bash
>> Proximity: HCSR04I2C - within
>> Error: NaN == 5.85
>> at Proximity.<anonymous> (test/proximity.js:1124:12)
>> at Proximity.<anonymous> (lib/mixins/within.js:31:18)
>> at Proximity.emit (events.js:107:17)
>> at Proximity.<anonymous> (lib/proximity.js:437:10)
>> at Object.sinon.clock.callTimer (node_modules/sinon/lib/sinon/util/fake_timers.js:238:32)
>> at Object.tick (node_modules/sinon/lib/sinon/util/fake_timers.js:183:28)
>> at Object.exports.Proximity: HCSR04I2C.within (test/proximity.js:1128:16)
```

I can't for the life of me see why it would be sending a NAN when the other two pass....

soundanalogous, 2015-11-11:
Is the backpack's I2C interface specific to the HCSR04? Would it be possible to create an i2c interface that covers a range of proximity devices (those that would require a backpack at least), or is the HCSR04 the only proximity sensor that currently needs a backpack?

ajfisher, 2015-11-11:
So I haven't tested this with a parallax Ping but it should work. At the moment, I think this is the only one that doesn't provide an i2c interface that is in common use within the nodebots world (as they are only a couple of bucks)

frxnz, 2015-11-13:
This was closed with https://github.com/rwaldron/johnny-five/commit/662098bce2efa52549e35fd387aeac7b711292ea

closed 2015-11-13 by frxnz