rwaldron/johnny-five #958: Update wiki API docs pages: Reformat "Shape" sections

rwaldron, 2015-11-10:
Here's an example of the new format: 

```
| Property Name | Description | Read Only |
|---------------| ----------- | ----------|
| `accelerometer` | An instance of `Accelerometer` class. | Yes |
| `altimeter` | An instance of `Altimeter` class. | Yes |
| `barometer` | An instance of `Barometer` class. | Yes |
| `gyro` | An instance of `Gyro` class. | Yes |
| `hygrometer` | An instance of `Hygrometer` class. | Yes |
| `thermometer` | An instance of `Thermometer` class. | Yes |
```

scottgonzalez, 2015-11-10:
Do you want a checklist to ease collaboration?

* [ ] Accelerometer
* [ ] Altimeter
* [ ] Animation
* [ ] Barometer
* [ ] Board
* [ ] Boards
* [ ] Button
* [ ] Compass
* [ ] ESC
* [ ] Expander
* [ ] Gyro
* [ ] Hygrometer
* [ ] IMU
* [ ] IR.Reflectance
* [ ] Joystick
* [ ] LCD
* [ ] Led
* [ ] Led.Digits
* [ ] Led.Matrix
* [ ] Led.RGB
* [ ] Motion
* [ ] Motor
* [ ] Multi
* [ ] Piezo
* [ ] Pin
* [ ] Ping
* [ ] Proximity
* [ ] Relay
* [ ] Sensor
* [ ] Servo
* [ ] ShiftRegister
* [ ] Sonar
* [ ] Stepper
* [ ] Switch
* [ ] Temperature
* [ ] Thermometer

rwaldron, 2016-01-22:
Th

closed 2016-01-22 by rwaldron