rwaldron/johnny-five #959: Motor: some commands skipped

cescopag, 2015-11-10:
Hi everyone, i'm having weird problems trying to control two motors with Arduino and J5.
I'm using a L293D chip as motor driver with external power input.
I'm testing this simple script:

    var motorA, motorB;
    var five = require('johnny-five');
    var board = new five.Board();
    board.on('ready', start);

    function start() {

    	motorA = new five.Motor({
    		pins:{
    			pwm: 6,
    			dir: 3,
    			cdir: 2
    		}
    	});
    	
    	motorB = new five.Motor({
    		pins:{
    			pwm: 9,
    			dir: 5,
    			cdir: 4
    		}
    	});
    	
    	board.repl.inject({
    		a:motorA,
    		b:motorB
    	});
    }

When I try to turn motors on/off, some commands are somewhat ignored.
For example, I've just tried to type these commands:

a.forward() -> the motor A starts correctly
b.forward() -> the motor B starts correctly
a.stop() -> motor A stops correctly
b.stop() -> nothing happens, motor B still running
Then, I pressed arrow up and the B motor suddenly stopped!

I tried several times with different sequences and it happens quite regularly. In particular, stop commands seems to work better than forward commands...
The weird thing is that Arduino RX led blinks correctly every time, but some messages are just ignored and received by Arduino only when I press any key (sometimes 2-3 keystrokes are needed).

I'm quite sure the hardware part is fine, I wrote a test program with just the Arduino IDE that uses raw pin commands and motors work fine.
The only difference I can think about is that in my test program I set the serial port at 9600 baud.

Any ideas on what's going on?

cescopag, 2015-11-10:
UPDATE: I tried setting firmata and johnny-five both to 9600 bauds but I got the same issues

rwaldron, 2015-11-10:
@dtex this sounds strangely like that thing we encountered at jsconf... Any thoughts?

dtex, 2015-11-11:
I'm trying to remember what we did to make it repeatable (if we ever did make it repeatable). It only happened on your computer right? I'm also trying to identify all the places where the bug could exist. What about hitting extra keys on the keyboard could possibly cause the value of a pin to change? That really sounds like something in USB/Serial/hardware land.

@cescopag I have lots of ideas to help isolate the problem but I don't want to waste your time. I'm going to try to replicate on my end. I know it's an L293D based board, but which board is it? Are you on Windows? Rick and I were seeing weirdness on something other than the L293D so I don't think this is specific to your motor controller. I also don't think this is Johnny-Five but I need to be able to replicate to prove that.

dtex, 2015-11-11:
... but it was with Motor() both times and I haven't heard of anything like this with any other class. LED is pretty much the same thing so...

cescopag, 2015-11-11:
Thank you for your feedback!

@dtex I'm using a proto-shield I built myself, but it's just a L293D pre-patched to some pins. It is basically equivalent to this: https://csacomani.files.wordpress.com/2012/09/l293d_proj.jpg

I'm using a Macbook and Arduino UNO via USB. I was using a HC-06 bluetooth adapter in the past, and I got similar problems. I'm not sure it was the exact same behavior, but I guess it's not the USB itself.

I'm going to do some deep testing this weekend, I'll keep you updated. Your ideas would be very appreciated!

dtex, 2015-11-11:
This is silly and is almost certainly a dead end but it's easy to check. Exactly which Macbook and USB host controller are you using? I wonder if they match @rwaldron's. 

These are the items I'm talking about:
[image]

Assuming this isn't the culprit (and it probably isn't) I'd proceed to checking the call stack from motor.stop all the way down to Serialport sending the message with a good ol' fashion set of console.logs and see where the calls stop. My gut says this is either in the REPL or Firmata.

dtex, 2015-11-11:
I made this little test rig using LED's, but still using all your code with motor() controlling the pin values and I was unable to replicate:

[image]

Expected behavior is:
````
// both green lights turn on when program starts
a.forward(); // one of the yellow lights comes on at 50%
b.forward(); // the other yellow light comes on at 50%
a.stop(); // The first yellow light stays on
b.stop(); // The second yellow light stays on
````
We can eliminate the motor controller if the behavior persists for you with this setup.

dtex, 2015-11-11:
Heck, you don't even need the red wires and LED's to test that, just green and yellow.

cescopag, 2015-11-11:
That was a very simple but clever test! I just tried it and made a video, so you can see the problem occurring a couple of times:
https://www.youtube.com/watch?v=fqh6WpT-zW4

Here is the details of my mac:
[image]

dtex, 2015-11-12:
The mac details didn't come through but the video did and that's some seriously weird shit. I don't know what to think. We know the calls are firing because the motor object is returned and written to the console. Tapping "W" on the keyboard a couple of times would not cause anything to happen in johnny-five, firmata, serialport or Arduino land but it seems to shake the commands loose somehow. They have to be getting stuck in USB land whatever that means.

Would you mind hacking on Firmata a little?

In ```/node_modules/firmata/lib/firmata.js``` find these lines:

````
Board.prototype.analogWrite = function(pin, value) {
  var data = [];
````
and change them to:
````
Board.prototype.analogWrite = function(pin, value) {
  console.log("AnalogWrite", pin, value);
  var data = [];
````

and then these lines:
````
Board.prototype.digitalWrite = function(pin, value) {
  var port = Math.floor(pin / 8);
````
and change them to:
````
Board.prototype.digitalWrite = function(pin, value) {
  console.log("DigitalWrite", pin, value);
  var port = Math.floor(pin / 8);
````
and then run your program on the test rig. I would like to know if the console.logs happen before you tap, tap, tap on the keyboard to unstick the commands. If they do then we know the problem is not in Johnny-Five and we can focus on Firmata, Serialport and the USB controller.

Thank you so much for testing this stuff. I'm dying to figure out what's going on here.

cescopag, 2015-11-12:
Ok it prints both log lines immediately after each command every time, and nothing when I tap on the keyboard...

In this screenshot the a.forward() didn't turn the led on, but you can see the logs just after the a.forward()
https://www.dropbox.com/s/5leyj8nxx8v7tzz/Screenshot%202015-11-12%2007.54.46.png?dl=0

dtex, 2015-11-12:
So it seems to me our problem has to be in node-serialport or the USB Controller. I wonder if sending four commands in such short order is contributing to the problem...

Wait a minute.... did you try a different USB cable?

cescopag, 2015-11-12:
yes, tested with two two cables, same problem...

cescopag, 2015-11-12:
just tried with a linux laptop and it works like a charm!

dtex, 2015-11-12:
Ya, it works on other Macs as well. 

I just learned a trick that will help us eliminate firmata and the JS side of node-serialport. It's like the firmata hack but you don't have to edit anything. If you normally run your program like this:
````
node test
````
try this instead
````
DEBUG=serialport node test
````

It's kind of a pain because it outputs the motor object every time you call a method but if you scroll up you should see something like:
````
a.forward()
serialport Write: {"type":"Buffer","data":[230,0,0]} +18s
serialport Write: {"type":"Buffer","data":[144,40,0]} +0ms
serialport Write: {"type":"Buffer","data":[144,40,0]} +1ms
serialport Write: {"type":"Buffer","data":[230,0,1]} +0ms
````

dtex, 2015-11-12:
Also, thanks for indulging me yet again and also thanks for noticing the traffic on the RX LED.

cescopag, 2015-11-12:
you're welcome, now I'm curious as well! here is the output, looks exactly the same as yours:
    
    >> a.forward()
    AnalogWrite 6 0
    serialport Write: {"type":"Buffer","data":[230,0,0]} +4s
    DigitalWrite 2 0
    serialport Write: {"type":"Buffer","data":[144,40,0]} +1ms
    DigitalWrite 3 1
    serialport Write: {"type":"Buffer","data":[144,40,0]} +0ms
    AnalogWrite 6 128
    serialport Write: {"type":"Buffer","data":[230,0,1]} +0ms

cescopag, 2015-11-12:
I was thinking... is there a way to "ping back" from arduino back to mac what it is receiving?

dtex, 2015-11-12:
And thats an example where it didn't respond right?

>  is there a way to "ping back" from arduino
I don't know to be honest. It would seem to me if we aren't seeing activity on the RX light when commands fail then the commands are not making it to the Arduino at all.

cescopag, 2015-11-13:
Yes that was when not responding...

I see what you say... Let's take the video test, I'm pretty sure this is what's going on:

1) Arduino received just the first two instructions, nothing happens (both pins set to zero)
2) When I type the first "w" the third instruction is sent, pin 3 turns on (but it's already on, green led)
3) Then I type the second "w", the fourth instruction is sent, pin 6 turns on and the yellow led lights up)

My question is, why are they sent after a keystroke? Who's taking care of that? is there a queue of serial commands somewhere?

dtex, 2015-11-13:
Oh man, I don't really know. These aren't things I've explored before.

Here's what I do know:
* All the commands are being sent from Johnny-Five to Firmata
* All the commands are being sent from Firmata to node-serialport
* Somewhere between the node-serialport javascript layer and the computer's USB port messages are getting stuck
* The messages are not lost, just stuck. Tapping any key on the keyboard unsticks them.
* This is happening on very few computers (maybe 2 that we are aware of)

Here's what I think I know:
* This is manifesting in motor because we are sending four commands over serial in very short order whereas other classes only need to send one... **I think this is an important clue**

Here are other things I don't know:
* Are the commands getting from the JS layer to the C layer in node-serialport?
* Are the commands getting from the C layer to the OS interface?
* Are the commands getting from the OS interface to the driver?

There are some smart people* who know more than I do and can help us here, but we're going to need to be able to recreate the issue on other computers. To do that, I think we should get all the details on your setup. What versions of each of these are you running:
* Johnny-Five
* Firmata
* node-serialport

Which OS, mac model and USB driver? (that screenshot never came though)

Is there anything extra about your setup (USB hubs, plugged in through keyboard instead of USB port, etc)?

@rwaldron Do you have time to try out @cescopag's code with the rig I fritzed up above? I wonder if it's happening for you too.

* Those smart people I was thinking of are @voodootikigod and @jacobrosenthal

rwaldron, 2015-11-13:
[image]

Sorry I'm late with this...

cescopag, 2015-11-14:
this is my setup:
![mac details]
(https://dl.dropboxusercontent.com/u/218685/J5-problem/mymac.png)
no USB hub or keyboard... just arduino straight into usb port

cescopag, 2015-11-14:
node.js version is 0.12.4
how can I find out the version of other parts?

dtex, 2015-11-14:
Your hardware is very different from @rwaldron's so likely nothing there. I'm going to install those versions and see if I can replicate. I'm afraid it won't be until Monday or Tuesday night because I am at Nodevember and left my test rig at home.

dtex, 2015-11-23:
@cescopag I'm sorry it took me so long to get back on this. J5@0.8.100 is not the latest. Is there a reason you need to use that version? Can you test with J5@latest and see if it still happens? I honestly don't think it will change anything, but it's an easy thing to try.

cescopag, 2015-12-02:
@dtex I deteled node_modules, updated node.js to version 5 and reinstalled everything and it works perfectly! I don't know why I didn't try this before...

rwaldron, 2015-12-02:
@cescopag can this issue be closed?

closed 2015-12-02 by cescopag

dtex, 2015-12-02:
@cescopag Thanks for the update.

@rwaldron I reviewed all the commits between 0.8.100 and latest (as of 10 days ago) and there was nothing in our logs that would have fixed this. I suspect it was a node serialport issue but I really don't know for sure.

dtex, 2015-12-02:
Also, I'd be curious if your issue from JSConf has also been mysteriously fixed.

rwaldron, 2015-12-02:
@dtex I'll try to repro, but it might be a few days before I have a chance