rwaldron/johnny-five #966: Flashing EV3

iammerrick, 2015-11-19:
Would be great to have some documentation on how to flash the EV3 brick.

rwaldron, 2015-11-20:
I'm not sure I understand why? Johnny-Five code doesn't run on the EV3 Brick yet, that's a work in progress for 2016. The current Lego EV3 and NXT support is for use with the EVShield. http://johnny-five.io/news/lego-evshield-support/

iammerrick, 2015-11-20:
Oh I'm sorry @rwaldron I misunderstood. Thanks for the kind reply, sorry for the inconvenience.

closed 2015-11-20 by iammerrick

rwaldron, 2015-11-20:
No inconvenience at all! 

I _have_ been working on an IO plugin for running code directly on an EV3, but I have touched it in many months. I'm shooting to have something useable in spring of 2016

iammerrick, 2015-11-20:
Sweet! Looking forward to it!