rwaldron/johnny-five #967: Controlling Arduino Mega Ramps 1.4 using Johnny-five

phuwin, 2015-11-24:
I want to control a Arduino Mega with a Ramps 1.4 shield using Johnny-five.
I uploaded the Standard Firmata the Arduino board and tried to run this code:

    var five = require('johnny-five');
    var board = new five.Board();
    board.on('ready',function(){
     console.log('Board is ready');
    });

My error:

    1448365609699 Device(s) COM3,COM5
    1448365609707 Connected COM3
     1448365619710 Device or Firmware Error A timeout occurred while connecting to the Board.
    Please check that you've properly flashed the board with the correct firmware.
    See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting
    enter code here
    events.js:146
      throw err;
      ^
    Error: Uncaught, unspecified "error" event. ([object Object])
    at Board.emit (events.js:144:17)
    at Board.log (C:\Users\Digital Hammer\Documents\Electric lab         playground\test\node_modules\johnny-five\lib\board.js:633:8)
    at Board.(anonymous function) [as error] (C:\Users\Digital Hammer\Documents\Electric lab playground\test\node_modules\johnny-five\lib\board.js:644:14)
    at Board.<anonymous> (C:\Users\Digital Hammer\Documents\Electric lab playground\test\node_modules\johnny-five\lib\board.js:414:14)
    at Timer.listOnTimeout (timers.js:92:15)

I notice that my board is connected to COM5, but the connected one is COM3.
Does anyone have any ideas why it does not work?

closed 2015-11-24 by phuwin

rwaldron, 2015-11-24:
That error generally occurs when there is no valid Firmata on the board. 

```
1448365609699 Device(s) COM3,COM5
1448365609707 Connected COM3
```

> I notice that my board is connected to COM5, but the connected one is COM3.

I don't understand what you mean? The output: 

```
1448365609699 Device(s) COM3,COM5
```

Means that two devices have been detected.

```
1448365609707 Connected COM3
```

Means that the COM3 was connected. 

If you need to connect an explicit port, just specify it: 

```js
var five = require('johnny-five');
var board = new five.Board({
  port: "COM5"
});
board.on('ready',function(){
  console.log('Board is ready');
});
```

Hopefully that helps :)

phuwin, 2015-11-25:
@rwaldron that really helps, thanks a lot!