rwaldron/johnny-five #968: Why there is no Enable pin for Stepper?

phuwin, 2015-11-24:
five.Stepper({
       type: five.Stepper.TYPE.DRIVER
       stepsPerRev: number,
       pins: {
         step: number,
         dir: number
      }
     });

dtex, 2015-11-24:
I'm not too familiar with stepper but I suspect in most cases the enable pin is wired directly to logic v+ so it is always enabled. This saves pins on the Arduino.

I see in the example over at Johnny-Five.io that is not the case. In fact, enable is not wired to anything. That may just be an oversight in the diagram.

Tagging @dwayn because he might be able to help.

rwaldron, 2015-11-24:
There's no `enable` pin exposed in the underlying protocol. Let's ask @soundanalogous

soundanalogous, 2015-11-24:
@phuwin, which stepper motor driver are you using that requires an `enable` pin?

rwaldron, 2015-11-24:
@soundanalogous in another issue, @phuwin mentioned this: http://reprap.org/wiki/RAMPS_1.4, so possibly the same case here

soundanalogous, 2015-11-25:
It has an `enable` pin. However my recommendation is to use `digitalWrite` to set or clear the `enable` pin in the user's code rather than adding it as an option in the Stepper module.

phuwin, 2015-11-25:
@soundanalogous Thanks for your suggestions.
I tried to set the enable pin, but it still does not work.
````
var five = require('johnny-five');
var board = new five.Board({
	port:"COM5"
});

board.on('ready',function(){
	console.log('Board is ready');
	this.digitalWrite(38,0);
	var stepper = new five.Stepper({
		type: five.Stepper.TYPE.DRIVER,
		stepsPerRev: 200,
		pins: {step:54,dir:55}
	});
	this.digitalWrite(38,1);
	stepper.rpm(180).ccw().step(2000, function() {
		console.log("Done");
	});

});
````

It prints out 

````
Board is ready
Done
````

It would be nice if you can check my settings:
[![enter image description here][1]][1]<br>AC to DC converter
[![enter image description here][2]][2]<br>The board, I tried to plug the wires to 5V as well, but it did not work either.

How do I wire it to + or -? I am sorry for my ignorance about this.
One more thing is that my arduino is a clone one, would that be the issue?

  [1]: http://i.stack.imgur.com/arPcW.jpg
  [2]: http://i.stack.imgur.com/hWy0h.jpg

phuwin, 2015-11-25:
The stepper works without enable pin manipulated from the code in this case. But thank you all for the knowledge.

closed 2015-12-18 by dtex