rwaldron/johnny-five #969: Led.Digits not working with my 8-Digit LED Display Segment

idophir, 2015-11-26:
I've spent a few hours on this and I can't get my 8-Digit LED Display Module to work. No matter what character I try to draw or print it just prints random illegible characters or nothing.

I purchased my board in China but it works fine with the snippet below. What am I doing wrong?
Is there a way to send arduino code to the board from within the johnny-five project?

```
// Pin numbers
int rck = 4; // clock
int sck = 3; // cs
int dio = 2; // data

// Binary for 0 to 9
byte num[10]={B11000000, B11111001, B10100100, B10110000,
B10011001, B10010010, B10000010, B11111000, B10000000, B10011000};

// Positions
byte pos[8]={1, 2, 4, 8, 16, 32, 64, 128};

void setup() {
  pinMode(rck, OUTPUT);
  pinMode(sck, OUTPUT);
  pinMode(dio, OUTPUT);
}

void loop() {
  for (int j = 0; j < 8; j++) {
    digitalWrite(rck, LOW);
    shiftOut(dio, sck, MSBFIRST, pos[j]);
    shiftOut(dio, sck, MSBFIRST, num[j]);
    digitalWrite(rck, HIGH);  
    delay(2); // This keeps the flicker above 60 Hz
  }
}
```

rwaldron, 2015-11-26:
Can you link me to the device you're using (so I can acquire one)?

idophir, 2015-11-26:
Thanks for the quick reply. 
This is where I purchased it from: https://item.taobao.com/item.htm?spm=a1z09.2.0.0.AGtr6h&id=14526509137&_u=q20t89f44be2

I currently live in China so it was very easy to get. 
I have a feeling the display is expecting an 8-bit value while Led.Digits sends it a binary. Could that be the case?

rwaldron, 2015-11-26:
No, that wouldn't be the case. I have several different 8 digit displays that have been tested with Johnny-Five, but not this _exact_ unit. I'll figure out getting one of these, but in the meantime, can you give me two things?

1. Paste the _exact_ code that's failing
2. Post a picture of your hardware 

I'll do my best to help you as immediately as possible, but today is a holiday in the US and I will be AFK for the day in about an hour.

idophir, 2015-11-26:
This code makes the display show "8." on the leftmost position.

```
var five = require("johnny-five");
var myBoard;
myBoard = new five.Board();

myBoard.on("ready", function() {
	var display = new five.Led.Digits({
		pins: {
			data: 2,
  		    cs: 3,
		    clock: 4
		}
	});

	display.clear();
	display.draw(0, 1, 0x00);
});
```

Here is a picture of the display:
[image]

Thanks a lot for helping. Enjoy your holiday.

rwaldron, 2015-11-26:
`draw` method is for matrix devices (long story about how they all came to have the same backing class). Try this instead:

```js   
display.print("88888888");
```

rwaldron, 2015-11-26:
Also, need better docs :/

idophir, 2015-11-26:
Well, something has changed - now I see the numbers light up for a very short time from left to right. It does that twice and then shows a constant dot on the rightmost digit...

rwaldron, 2015-11-26:
Strange! I'm AFK now, but I can dig into this tomorrow—sorry for delaying your project :/

dtex, 2018-01-11:
Hi @idophir ,

Johnny-Five contributors and maintainers are loathe to close issues where someone has a need. We don't want anyone to ever feel that we don't care, but we just haven't gotten to this feature yet. Rather than leave it languishing as an open issue we have created a [Requested Features](../wiki/Requested-Features) page and added your request for the 595 there.

closed 2018-01-11 by dtex